EdgeTX/edgetx #1036: Preflight Checklist Switch Positions are not saved

PSi86, 2021-11-12:
On the model setup page under section "Preflight Checklist" you can select which switches should be checked by the radio.
Also the switch positions can be saved, but this is not happening.

**To Reproduce**
-Set switches to desired positions
-Go to Model Setup
-Scroll to the Switch Positions and activate the item (click the wheel).
-Scroll left to select all switches and click again.
-The arrows next to the switches now reflect the current positions of each.
-Exit the menu
-Load another model from Model List (or reboot radio)
-Go back to the model where you did the change.
-Check the preflight checklist switch postions - they are reset back to default.

**Expected behavior**
The switch positions should be saved to the model and retained on model change or reboot.

**Screenshots**
[image]

**General information**
 - nightly b4903d8
 - Frsky X9D+ 2019 (Access Version)

PSi86, 2021-11-13:
Wow, this was quick! Will test with the next nightly once the commit is merged and report back here.

raphaelcoeffic, 2021-11-13:
> Wow, this was quick! Will test with the next nightly once the commit is merged and report back here.

Normally we test *before* we merge. This is in fact a condition ;-)

PSi86, 2021-11-13:
Will you supply a build for testing?

raphaelcoeffic, 2021-11-13:
> Will you supply a build for testing?

It's in the PR. Head to "Checks", then you should see "Artifacts" at the bottom of the page.

PSi86, 2021-11-13:
Thank you, that was NOT easy to find :)
Testing is done -> works almost perfectly.
What does this mean?
After installing the build all switch position checks were disabled in all models.
You then have to re-enable every single switch by clicking on each one and then again select all of them and click to store all current switch positions like you normally would.
I did not test the behaviour when models are converted from opentx but coming from an older EdgeTX version it would be better to have all switch checks in the default positions rather than having all checks disabled.

[image]

What do you think?

raphaelcoeffic, 2021-11-13:
The problem here is that indeed, the positions were stored the wrong way. So your files were already borked. I tested the conversion from OTX and from ETX 2.5 and they seem fine.

PSi86, 2021-11-13:
Ok, then: Case closed! Nice work!

raphaelcoeffic, 2021-11-13:
@PSi86 issue are closed automatically when the code is merged ;-) we need @pfeerick to give it a hard time first 😋

PR #1040 (2021-12-17): Fix switch warnings
Fixes #1036 

Summary of changes:
- removed MENUS_LOCK (legacy)
- unified switch warn states to be the same on all targets
- fixed conversions from 2.3/2.4
closed 2021-12-17 by raphaelcoeffic