EdgeTX/edgetx #1044: Image/Media Viewer for EdgeTX

ulph0, 2021-11-13:
Hi, i would like to have a layouted image with all stick commands on a button of my radio. I can play a song with the special functions but this i can't. Of course it would be also great to have other Formats like SVG, markdown Text, PDF for other purposes (ie. view manuals or read a chapter of "don't panic" when your drone crashed :).

raphaelcoeffic, 2021-11-13:
> Hi, i would like to have a layouted image with all stick commands on a button of my radio. I can play a song with the special functions but this i can't. Of course it would be also great to have [...]

Please note that this is a microcontroller running on your radio, not a full-blown CPU like in you phone ;-) Once this is case, then we can talk about SVG, Markdown and PDF. Until then, simple pictures will have to be enough ;-)

closed 2021-11-13 by raphaelcoeffic

ulph0, 2021-11-13:
yeah simple pictures are the topic of this feature request, anything else was only nice to have

raphaelcoeffic, 2021-11-13:
> yeah simple pictures are the topic of this feature request, anything else would only nice to have. but why did you close it then?

You can already view pictures and have model notes, so no problem here.

ulph0, 2021-11-13:
oh ok! i'm sorry then, looked for it but didn't find, would you tell me what i have to do put that on a switch or which special function i have to use to show an image?

pfeerick, 2021-11-14:
Hm... you have us on that point... there's not 'show image' SF... but at the same time, you have standalone lua apps, and can have a lua app that shows all your stick positions, etc. In fact, you may actually be interested in the [Switch2 ](https://www.rcgroups.com/forums/showpost.php?p=47009445&postcount=3793) widget for that - you can set icons to indicate what each switch position does, and have it full screen on one of your screen layouts ;)

ulph0, 2021-11-14:
hi, thank you! i will look at that! on discord, i talked with sb who said it also would be possible if i do my own lua script widget.

ulph0, 2022-04-03:
just for people who accidentally fiend this closed thread, here is the image viewer widget script: https://www.schleth.com/fpv/vu-a-simple-image-viewer-for-edgetx-radios-with-big-screens-2113.html