EdgeTX/edgetx #1045: NV14 Curves window when trying to set a value, CV1, crashes the transmitter, restart required

Outwide, 2021-11-13:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version (e.g. 2.4.0 or nightly and #commit number)
 - Hardware (e.g. Frsky X9D+ or RM TX12S)

Outwide, 2021-11-13:
NV14 Setting Curve values in the main window. Smooth selected, In Type field change Standard to Custom the screen fades away and no visible screen available, restart required. Edge TX 2.5

After restarting go back to the Curve values in the main window. Smooth selected, In Type field, press 5pts. New window showing Curve to be adjusted opens. Press anything in the adjustment field at the bottom and transmitter crashes, screen fades away again and no visible screen available, restart required. Pictures attached. 
[image]
[image]
[image]

[image]
[image]

pfeerick, 2021-11-14:
This was fixed in the nighties about two weeks ago ;) https://github.com/EdgeTX/edgetx/pull/965

closed 2021-11-14 by pfeerick

Outwide, 2021-11-14:
I installed Edge TX 2.5 it yesterday afternoon.
I will do another clean install.

Thanks

Closed #1045<https://github.com/EdgeTX/edgetx/issues/1045>.

pfeerick, 2021-11-14:
You will either have to run a nightly, or wait for 2.6.0.