EdgeTX/edgetx #1046: Radiomaster TX16s Audio mute through special functions

onki69, 2021-11-14:
I want to add a BT audio module to my RM 16TX since there is just a speaker and no phone jack / BT audio availabe.
One problem is to mute the amplifier in order to just hear the voice output / vario through the BT audio.

There is a signal at the µC labelled Audio_mute  (at PA7 on the µC). It would be great to be able to switch that port through a special function using one of the default switches. Muting just menas to enable this signal in order to disable the speaker amplifier. The audio_out (at PA4 of the µC) should still output the audio signal to transmit it through BT.
The feature might also be helpful for people who want to add an phoe jack or even need a possibility to mute the speake for a short while without modifiing the speaker volume.

In addition it would be great to have the AUX1 (PA15) and maybe AUX2 (PB0) supply voltage switchable as well through special functions (to enable / disable the BT audio module) or other external stuff. AUX2 might be switched through the internal BT function as well.

onki69, 2022-04-10:
The new version of the Radiomaster TX16S comes with an audio phone jack.
Will there be a chance to use the mute signal?
Is the new version firmware compatiblke with the older hardware revision?

rotorman, 2022-04-10:
The new version is firmware compatible. When you plug headphones to the jack, the internal loudspeakers are silent. This is achieved electrically, without involvement of firmware.

pfeerick, 2022-04-12:
Even though it is currently being used to mute the amplifier when there is no audio playing, preventing continuous background noise, it shouldn't be too hard to add a flag those functions can check (which a SF - or perhaps a general option - i.e. under the sound options - mute switch option?) ... i.e. check the switch state before unmuting, mute when switch is set.

raphaelcoeffic, 2023-09-05:
Implemented in #3741

closed 2023-09-05 by raphaelcoeffic