EdgeTX/edgetx #1048: Yaml => Logical Switch Edge max period keeps getting set to `<<`

pfeerick, 2021-11-14:
It appears to be set during a session, but as soon as you select another model, or power cycle, it reverts to `<<`.

I.e. New model, with random Edge LS defined:
[image]

After swapping to another model and back:
[image]

ulfhedlund, 2021-11-18:
If I may add something similar: if I change the v1 value for an existing model on my TX16S, either up or down,  v2 goes from << to 39.5. After that if I change the v1, v2 changes as well. It is the same in both 2.5 and 2.6 (built today).
If I'm stating the obvious and already known - please just disregard my comment.

raphaelcoeffic, 2021-11-22:
@pfeerick isn't this fixed by #1113 ?

pfeerick, 2021-11-22:
I'll let you know once I've tried the nightly ;)

pfeerick, 2021-11-22:
Wooho! Yup, looks like I was right... #1110 was the same/similar issue ;)

closed 2021-11-22 by pfeerick

ulfhedlund, 2021-11-24:
Hi,
I received this message since I commented on issue 1048 a while back. I 
understand that issue is now closed, however the related issue that  I 
commented on still remains:

"If I may add something similar: if I change the v1 value for an 
existing model on my TX16S, either up or down, v2 goes from << to 39.5. 
After that if I change v1, v2 changes as well. It is the same in both 
2.5 and 2.6 (built today).
On my X-lite (on 2.5) it works as expected i.e. changing v1 doesn't 
effect v2."

I tested on my RM TX16S on a build on main branch that I just compiled a 
few minutes ago.

Should I add this as another issue or is this known and already being 
worked on?

Regards,
Ulf Hedlund

pfeerick, 2021-11-24:
That is a completely different issue - this was tracking nightly yml related problems. 

Keep in mind with the Edge SF, the second value for v2 (v1 is your switch/trigger) must always be bigger than the first, which is why it auto-increments if it is set to a value. Unfortunately, `<<` seems to be detected as a value also, and sets off the auto-increment as well.

ulfhedlund, 2021-11-25:
Ok, I understand. So it may be fixed in later versions, I suppose.
Thank you for all your efforts with EdgeTX, it is fantastic and I am very grateful.

Cheers,
Ulf