EdgeTX/edgetx #1050: Pot and slider warnings not properly read from radio in Companion

ulfhedlund, 2021-11-14:
The pot warnings set in a RMTX16S radio doesn't gets properly read when models and settings are read from the radio in Companion. They seem to get inverted. I have attached two pair of screen shots to illustrate what happens.

My environment: EdgeTX version 2.5, RadioMaster TX16S, Linux Companion (AppImage 2.5.0-28cdb40-x86_64).

[image]
[image]
[image]
[image]

pfeerick, 2021-11-15:
Confirmed with Companion 2.6, TX16S v2.5.0 firmware. TX12 (B&W radio) _appears_ to be ok.

elecpower, 2021-11-15:
Checked code and Companion does not differentiate between radios except for supported hardware when reading these settings so if as @pfeerick has advised "B&W appears to be ok" and as reported is inverted for TX16S (colorlcd) then suspicion moves to radio side. Can one of you radio experts please check?

ulfhedlund, 2021-11-16:
I can also tested on my X-lite BW radio and that seems to work as expected, so I can confirm what @pfeerick writes.

elecpower, 2021-11-16:
> I also tested on my X-lite BW radio and that seems to work as expected, so I can confirm what @pfeerick writes. The pot warnings seem to get inverted when writing to the TX16S radio from Companion as well. So it's both ways.

Consistency is good ;-)

elecpower, 2021-12-11:
So it would appear it is a bug that was introduced to the radio firmware rather than Companion?
Also if left the way it is then radio to radio conversions on both platforms are going to be affected and the YAML values would have to be interpreted inverse as well. Since the YAML files do not contain the radio that created then more issues.
Too much confusion.

raphaelcoeffic, 2021-12-14:
I might be wrong, but I think this is fixed in latest nightly build.

ulfhedlund, 2021-12-18:
A quick test I just did with a fresh build from today shows there are still problems with the TX16S. It seems the warnings are no longer reversed but they are not correctly transferred to/from radio and Companion, as far as I can tell.

PR #1981 (2022-06-07): Fix pot and slider warnings in Companion and radios
Fixes #1843
Fixes #1730
Fixes #1663
Fixes #1504
Fixes #1050

Summary of changes:
- fix Companion yaml model read and write
- B&W radios treat bit values the same as color to avoid unnecessary complication for Companion and radio conversions
- B&W radios default is therefore off for all pots and sliders
- B&W radios fix runtime warnings check
- all radios fix to save last pot or slider position when checked and mode AUTO on change of model or power off
- all radios do not display Pots Warning in model setup if no pots or sliders supported

Not [... cut]
closed 2022-06-07 by pfeerick