EdgeTX/edgetx #1051: Model => Center Beep does not beep or trigger hepatic

DavidLorentzen, 2021-11-14:
*Before submitting,  search the open issue if there is already a report of the bug*
Done, no report

**Describe the bug**
In the Model page, selecting a control for Center Beep does not produce a beep or a hepatic vibration. I believe this was broken between v2.4 and 2.5.0

**To Reproduce**
Tap to highlight control, does not trigger beep event.

**Expected behavior**
Expect audio and hepatic feedback when the selected control is centered

This issue has been confirmed by another user on RC Groups (see https://www.rcgroups.com/forums/showpost.php?p=48088265&postcount=1066)

**General information**
Tx16s, EdgeTx 2.5.0

eshifri, 2021-11-14:
It works in the current code.

lombartz, 2021-11-14:
... but not for the sliders on TX16S:

- the rectangle for the left slider ist empty
- both switches don´t highlight when toggling
- there`s no center beep or haptic on both sliders

TX16S, latest nightly

eshifri, 2021-11-14:
Works for me. The only (known) problem is with highlight (#411 ).

MervDale, 2021-11-15:
Same issue on latest nightly Eachine TX16S  

the rectangle for the left slider is empty
neither switches  highlight when toggling
there`s no center beep or haptic on both sliders

pfeerick, 2021-11-15:
> there`s no center beep or haptic on both sliders

I have to contradict you there, I'm afraid... if I disregard the highlight, and press the LS and RS center beep buttons (toggling them on, but not visibly so), the sliders do beep at center - as long as you don't sweep them too fast that they pass center too quickly.

MervDale, 2021-11-15:
Sorry but in my case on the Eachine  testing several different models, they do not work no matter what you do and they don't highlight as well as the L character is missing

pfeerick, 2021-11-15:
They'll highlight when you press them (i.e. indicating the buttons have focus), but don't indicate the actual active or inactive state :-/ https://youtu.be/tEifsC2eKYU

DavidLorentzen, 2021-11-15:
Here's a video. Radio volume is on full, and hepatic should also be heard in the video if it were to be triggered.

https://www.youtube.com/watch?v=Zd-TpejFizg

eshifri, 2021-11-15:
Mystery... :-(

DavidLorentzen, 2021-11-15:
I started wondering if it was a hardware issue, so I reloaded OpenTx on the same transmitter. Works. See video.

https://www.youtube.com/watch?v=MMd3RltOxFE

eshifri, 2021-11-15:
The very suspicious thing is that the letter "L" is gone.

lombartz, 2021-11-15:
If I set EX1 to EX4 to"Pot" with detent in Radio settings i get the following options in Model settings:
R - E - T - A - 1 - 2 - 4 - 5 - L - R - (empty) - R
Toggling with highlighting then only works on:
R - E - T - A - 1 - 2 - 4

eshifri, 2021-11-15:
Do you have FlySky sticks? If you do no, can you please try setting all EX to "None"?
There is definitely a problem wit setting active state (yellow) for some buttons in the second row, but beep works for me in all cases.

raphaelcoeffic, 2021-11-15:
> Do you have FlySky sticks? If you do no, can you please try setting all EX to "None"? There is definitely a problem wit setting active state (yellow) for some buttons in the second row, but beep works for me in all cases.

Seems the EXn are quite messing the whole thing.... Needs repair.

eshifri, 2021-11-15:
Yes, on my TX16 EX1 and EX2 were set to None and they were shown (and worked) on the model setup screen.
When EX1 and 2 are set to Pots, EX4 an 5 appear on the setup screen (but on the HW screen).
That being said, BEEP works correctly for me in all cases.

MervDale, 2021-11-15:
I have a bigger problem with the sliders, I noticed the indicators were not moving when I operated them so I  recalibrated the  the radio I now have indicators  that the show them working and I can turn them on and off but instead of getting haptic I get   L =TIMER 1 Elapsed  R= Timer 2 Elapsed  on all models and nowhere in my programs do I call the L or R sliders at all ?   Reinstalled 2.5 and all is well sliders included working as they should... put the nightly back on and back to slider vocals    going back thru nightlies this started with tx16s-493c90d   prior to this is ok on my Eachine TX16S (edited
[YMLfiles.zip](https://github.com/EdgeTX/edgetx/files/7542305/YMLfiles.zip)
)

DavidLorentzen, 2021-11-16:
This is not a "slider thread". The topic is center beep.

MervDale, 2021-11-16:
Well sorry about that, some own else raised the issue with the sliders  center beep,  not me,  I just carried on.

DavidLorentzen, 2021-11-17:
I identified the problem, and of course it was the incomplete understanding of the user.

In the "Radio Setup", I didn't know about the "Mode" setting for the Alarm and Hepatic. Both were set to "Alarm" in my EdgeTx default installation, or the translation of my settings from OpenTx to EdgeTx, because I frankly didn't know about it enough to have manually changed it. Setting both to "All" triggered the sound and beep on "Center Beep" setting for the model.

This does exactly what I want to be able to do; that is, have hepatic without a beep on "Center Beep".

Sorry if this confused anyone, and I hope it clarifies the issue for anyone having this problem.

-= Dave

closed 2021-11-17 by DavidLorentzen

pfeerick, 2021-11-18:
Ah, derp... didn't think to ask about the sound mode setting... Glad it works as it shoud for you... now it's just that pesky indicator button highlight to figure out! :)

eshifri, 2021-11-18:
Actually, I'm still seeing problem with sliders in the after YAML introduction SW:
1. Letter "L" for left slider is not displayed (empty rectangle).
2. Beep does not work for both sliders.
And, BTW, the highlighting is shown correctly if you exit the setup screen and return back. I thought it was some compiler over-optimization and created a variable to held the return of the bfSingleBitGet() - did not make any difference.

MervDale, 2021-11-18:
I think you will find this has been moved to #1062

eshifri, 2021-11-18:
Hm, may be. But I do not get Timer elapsed. :-)
Anyway, I think you are right that this does not belong here.