EdgeTX/edgetx #1053: Receiver ID

MuhammadQaisarAli, 2021-11-14:
**Problem**
When we bind a new model, the receiver ID always starts from 0, unlike OpenTx, EdgeTx won't make a warning of an ID, if already being used, which makes it very easy to overwrite the old model receiver ID. 

**Suggestion**
If possible please provide a dropdown list while selecting receiver ID, the elements of the list should have revived ID, and a corresponding model name if used, with the help of this, it would be very easy to select a unique ID for each receiver/Model.

e.g.

0:Moldel1
1:Quad
2:Plan
3:Not Used
4:Not Used
.
.
.

**General information**
 - EdgeTX version: 2.5.
 - Hardware: RM TX16S)

raphaelcoeffic, 2021-11-14:
The main issue here is that it makes it necessary to scan all the model files to make it accurate. We will try to find another way to make it correct, but that might take a bit of time.

Eldenroot, 2021-11-14:
Why works in opentx and not here? why not to use same approach?

MuhammadQaisarAli, 2021-11-14:
> The main issue here is that it makes it necessary to scan all the model files to make it accurate. We will try to find another way to make it correct, but that might take a bit of time.

No worries, sir.. take your time.. we will wait...
Best Regards..

pfeerick, 2021-11-16:
> Why works in opentx and not here? why not to use same approach?

Would you like me to write a short novel on that? :grin: 

Anyway, it falls broadly down to two aspects - the UI has been completely re-implemented, so there may be things missing. But more importantly, the model storage model is being completely re-written, and now models have their own configuration files... and given the receiver ID is stored in each models file, this means you have to parse every single model file to see if there is a match... which is going to take time. So we need to figure out a mechanism to do this, which isn't going to make the UI unresponsive whilst doing so. 

I think when we were discussing it at one point, I may have suggested a button like "Next available ID" which you could press, thus it's a user initiated delay with corresponding UI "please wait" type thingy.

Eldenroot, 2021-11-16:
OK, thank you for your reply.

Maybe would be nice to track all missing features which were lost during conversion to new UI.

Wikiwix, 2022-10-16:
[Version 2.8.0, currently in RC stage](https://github.com/EdgeTX/edgetx/releases/tag/v2.8.0-rc3) does address this issue to some extent:

> When setting model/receiver IDs, the firmware will now tell you if the number is unique (not already used) or if it is in use by another model (and which!!) - and better than that, only IF it is using the same RF protocol (since you don't care about the ID of a model using a different protocol).

I believe https://github.com/EdgeTX/edgetx/pull/1374 is the PR adding this functionality

pfeerick, 2022-10-16:
Indeed, I believe the core issue is fixed (and yes, by that PR) as it not only tells you that there is a model using that ID, but also shows you want models the clash is with. Thanks! :)

closed 2022-10-16 by pfeerick