EdgeTX/edgetx #1061: nightly latest f1baa38

pds59, 2021-11-15:
radiomaster t16s:  updated to latest nightly version all models are gone. i try to upload with edge compagnion my latest b.u. file but no models are available if i go back to latest main firmware all model are available again

pfeerick, 2021-11-16:
Since you are on the nightlies, and yaml is in effect, after reloading from companion, you will probably need to delete the `RADIO/radio.yml` file so that the changes are re-imported.

pds59, 2021-11-16:
Ok i Will tryPaolo de scaglia

pds59, 2021-11-16:
Working tks

paolo

 

Habitatika di Paolo de Scaglia

Via del Salice 1 

28836 Vezzo VB

Cell. +39-3466393930

pfeerick, 2021-11-16:
Excellent. When you did the update, what version of EdgeTX were you updating from - i.e. was it a release version, a nightly, or something else?

pds59, 2021-11-17:
Hi, it was latest  firmware main in compagnion

 

Habitatika di Paolo de Scaglia

Via del Salice 1 

28836 Vezzo VB

Cell. +39-3466393930

pfeerick, 2021-11-18:
? Do you mean you updated from OpenTX 2.3.14 directly to a EdgeTX nightly?

pds59, 2021-11-18:
Hi, no i was on edge tx main latest version to edge tx nightly

 

Habitatika di Paolo de Scaglia

Via del Salice 1 

28836 Vezzo VB

Cell. +39-3466393930

closed 2021-11-22 by raphaelcoeffic

pds59, 2021-11-22:
Tks a lot and tks for flysky gimbal autodetect you dev are doing an incredible work I hope people will understand about the work

Tks again

Best regards

paolo

 

Habitatika di Paolo de Scaglia

Via del Salice 1 

28836 Vezzo VB

Cell. +39-3466393930

 

Closed #1061 <https://github.com/EdgeTX/edgetx/issues/1061>  via #1127 <https://github.com/EdgeTX/edgetx/pull/1127> .