EdgeTX/edgetx #1063: PR#1000 broke simu touch support

pfeerick, 2021-11-16:
@gagarinlg It looks like #1000 (or more likely the matching libopenui PR) broke `simu` touch support and unfortunately I didn't catch it during testing since I was focused on on-hardware testing 😮 I'm not sure if it flows through to the main Companion simulator as well, or only affects `simu` ... will have to investigate further.

gagarinlg, 2021-11-16:
> @gagarinlg It looks like #1000 (or its matching libopenui PR) broke `simu` touch support and unfortunately I didn't catch it during testing since I was focused on on-hardware testing 😮 I'm not sure if it flows through to the main Companion simulator as well, or only affects `simu` ... will have to investigate further.

I will test again in the evening.

gagarinlg, 2021-11-16:
> I will test again in the evening.
I tested with current main, TX16s and NV14 simulator.
I stared the simulator from companion and command line, everything worked fine using mouse and a touch screen as input devices. What exactly happened in your test?

dlktdr, 2021-11-16:
I can confirm that simu isn't working with touch after https://github.com/EdgeTX/edgetx/commit/52013da098633eaed1c6c15b56adff0ae5ba5ff6

gagarinlg, 2021-11-16:
So, obviously I am missing something here. What is simu, if it is not the simulator?

pfeerick, 2021-11-16:
For what it's worth, this mostly fixes it... 

simu is another target, a much simpler simulator... avoids the need to build the entire simulator and libsimulator suite...

[image]

```diff
diff --git a/radio/src/simu.cpp b/radio/src/simu.cpp
index dcff7f081..99df475f0 100644
--- a/radio/src/simu.cpp
+++ b/radio/src/simu.cpp
@@ -271,12 +271,13 @@ long OpenTxSim::onMouseDown(FXObject *, FXSelector, void * v)
 
 #if defined(HARDWARE_TOUCH)
   now = get_tmr10ms();
-  touchState.tapCount = 0;
+  simTouchState.tapCount = 0;
 
-  touchState.event = TE_DOWN;
-  touchState.startX = touchState.x = evt->win_x;
-  touchState.startY = touchState.y = evt->win_y;
+  simTouchState.event = TE_DOWN;
+  simTouchState.startX = simTouchState.x = evt->win_x;
+  simTouchState.startY = simTouchState.y = evt->win_y;
   downTime = now;
+  simTouchOccured = true;
 #endif
 
   return 0;
@@ -292,22 +293,23 @@ long OpenTxSim::onMouseUp(FXObject*,FXSelector,void*v)
 #if defined(HARDWARE_TOUCH)
   now = get_tmr10ms();
 
-  if (touchState.event == TE_DOWN) {
-    touchState.event = TE_UP;
-    touchState.x = touchState.startX;
-    touchState.y = touchState.startY;
+  if (simTouchState.event == TE_DOWN) {
+    simTouchState.event = TE_UP;
+    simTouchState.x = simTouchState.startX;
+    simTouchState.y = simTouchState.startY;
     if (now - downTime <= TAP_TIME) {
       if (now - tapTime > TAP_TIME)
         tapCount = 1;
       else
         tapCount++;
-      touchState.tapCount = tapCount;
+      simTouchState.tapCount = tapCount;
       tapTime = now;
     }
   }
   else {
-    touchState.event = TE_SLIDE_END;
+    simTouchState.event = TE_SLIDE_END;
   }
+  simTouchOccured = true;
 #endif
 
   return 0;
@@ -322,13 +324,14 @@ long OpenTxSim::onMouseMove(FXObject*,FXSelector,void*v)
     TRACE_WINDOWS("[Mouse Move] %d %d", evt->win_x, evt->win_y);
 
 #if defined(HARDWARE_TOUCH)
-    touchState.deltaX += evt->win_x - touchState.x;
-    touchState.deltaY += evt->win_y - touchState.y;
-    if (touchState.event == TE_SLIDE || abs(touchState.deltaX) >= SLIDE_RANGE || abs(touchState.deltaY) >= SLIDE_RANGE) {
-      touchState.event = TE_SLIDE;
-      touchState.x = evt->win_x;
-      touchState.y = evt->win_y;
+    simTouchState.deltaX += evt->win_x - simTouchState.x;
+    simTouchState.deltaY += evt->win_y - simTouchState.y;
+    if (simTouchState.event == TE_SLIDE || abs(simTouchState.deltaX) >= SLIDE_RANGE || abs(simTouchState.deltaY) >= SLIDE_RANGE) {
+      simTouchState.event = TE_SLIDE;
+      simTouchState.x = evt->win_x;
+      simTouchState.y = evt->win_y;
     }
+    simTouchOccured = true;
 #endif
   }
 ```

pfeerick, 2021-11-16:
Sorry, I updated the comment... "For what it's worth, this mostly fixes it... (swipes seem to latch / are janky)."

pfeerick, 2021-11-16:
Sorry for the delay, full Companion simulator is fine, so only `simu` needs TLC.

gagarinlg, 2021-11-16:
I tested with you changes applied. For me it feels the same in simu and simulator, after applying the patch for simu.
One could argue, that it is generally not really nice, because, when you move into on direction and then reverse, you will continue to scroll into the first direction, until you reach your starting point and only the the scroll direction is reversed, instead of having an absolute scrolling position attached to your mouse position, based on the starting point, but this seems to me, like it has been so before the changes I made.
Currently I would consider this fixed with your changes and would opt for a new issue for a nicer scrolling behavior for the simulators.

pfeerick, 2021-11-17:
Sounds fine to me... we can live with touch swipe not working quite right as opposed to touch being completely MIA 😆.

More work is needed for both simu and simulator, as they both behave the same now AFAICT. This is not how it was before the PR, as slide worked perfectly in both simulators prior to this

i.e. this is simulator as it was before, and simu how it is after being patched. Anyway, this is subject for another PR. 

https://user-images.githubusercontent.com/5500713/142088408-5319b912-e6a7-4063-b9f3-453ecef5303e.mp4

.

closed 2021-11-17 by pfeerick

gagarinlg, 2021-11-17:
PR #1070 fixes the strange touch behavior

commit 15a1981d (2021-11-17): Fix simulated touch not working in `simu`
Resolves #1063