EdgeTX/edgetx #1065: Make the internal model number as used in model00.bin or model01.yml usable in LUA

wimalopaan, 2021-11-16:
Add a field model.getInfo().number that returns the number of the model used in building the .yml file

wimalopaan, 2021-11-17:
Maybe there is more useful information one needs in lua scripts. But the model-number is quite crucial to generate filenames that are unique on the sd-card (like the model-files model<number>.yml)

pfeerick, 2021-11-22:
I'm going to have a stab at this ... so would adding it as another return value for `model.getInfo()` be acceptable? Currently it returns the model name, and bitmap path... so seems like adding `modelId` as a third value is a good fit. 

I *think* this should do it...
```
 lua_pushtablenstring(L, "modelId", g_model.header.modelId);
```

wimalopaan, 2021-11-22:
Well, I saw this in the sources, but the comments say this is the receiver number ...

wimalopaan, 2021-11-22:
Maybe:

lua_pushtablenstring(L, "modelIfilename", g_model.header.modelFilename);

raphaelcoeffic, 2021-11-22:
This is what you are looking for: `g_eeGeneral.currModel`.

pfeerick, 2021-11-22:
Well, we can do the filename anyway, since it's 'model related'. Thank you for that hint re: `g_eeGeneral.currModel` ;)

PR #1132 (2021-11-24): Add filename to lua `model.getInfo()`
Proposal to add a `filename` table element to model.getInfo, so that the yaml filename can be used in a lua script.

Resolves #1065 

Tested as functional on TX16S (colourlcd) and TX12 (B&W).

@raphaelcoeffic I'm not particularly happy just copying and pasting the `getModelNumberStr` function in... is there a better way to do this?

Test script:
```lua
local w, h

local function run(event)
  lcd.clear()
  local info = model.getInfo()
  if LCD_H > 64
    w, h = lcd.sizeText("Model Info")
  else
    h = 8
  end

  lcd.drawText(1, 0, "Model Info")
  
  if inf [... cut]
closed 2021-11-24 by pfeerick