EdgeTX/edgetx #1067: Make the default setting for curves Y = X instead of Y = 0.

Miami-Mike, 2021-11-16:
This went stale after [6 months on the OpenTX forum](https://github.com/opentx/opentx/issues/8492#issuecomment-970224829) but I think it's a good idea:

Instead of the default for curves being Y = 0% regardless of X, make it Y = X. That is, instead of a horizontal line on the X axis, make it a diagonal line from (X = -100%, Y = -100%) to (X = +100%, Y = +100%). That way, if a curve is applied to an input, mix, or output without first defining it, it won't cause unintended model behavior such as a motor suddenly turning on. This is already what you get by clicking "Apply" with the Curve Creator at its default settings, so why not make it the curve default?

pfeerick, 2021-11-16:
Sounds like an excellent idea... the default "curve" really should be... well, normal stick movement... 🤦

raphaelcoeffic, 2021-11-17:
@Miami-Mike i had the feeling I had done exactly this already. But maybe this was in a dream.

Miami-Mike, 2021-11-18:
[You commented about it in the OpenTX thread](https://github.com/opentx/opentx/issues/8492#issuecomment-835947616) on May 9th.

pfeerick, 2021-11-18:
Wait a minute... "Curve creator" ... you're talking about Companion, not the radio, aren't you? As the radio firmware defaults to creating a -100 => 100 curve by default.

Miami-Mike, 2021-11-18:
I probably shouldn't have mentioned the Curve Creator because it really doesn't have anything to do with this.

If you make it so that a curve can't be selected until it's defined then that might solve the issue that chronowarper and Raphael mentioned, but it could be a bit inconvenient. If you're setting up an input, mix, or output and you want to include a curve, you might find it annoying that the system won't let you do that. You'd first have to interrupt what you're doing, go to the Curves page, define the curve, and then go back to where you were.

I think the way I suggested would work out better because a Y = X curve could be included in any input, mix, or output, but would have no effect. It would be essentially transparent. You'd be able to first finish setting up the line you're working on and then proceed to the Curves page to adjust the curve.

elecpower, 2021-11-18:
As an aside, the ability to edit a curve from a mix or input is on my long todo list

elecpower, 2021-11-18:
I can knock up the Companion PR for Y = X if we are in agreement. Thumbs up/down @raphaelcoeffic @pfeerick

Miami-Mike, 2021-11-23:
Don't forget that when you do _Print Model_ in Companion, curves with all Y=0 values are presumably considered unused and don't show up in the printout. You'll want to change that so that instead, curves with all Y=X values are the ones omitted from printouts.

elecpower, 2021-11-23:
Tested and Y=X now not shown

Eldenroot, 2022-10-08:
> Sounds like an excellent idea... the default "curve" really should be... well, normal stick movement... 🤦 ... if I get a chance in the next day or so, I'll look at it...

This means that the curve editing would be more newbie friendly?

pfeerick, 2022-10-09:
No, it has nothing to do with that. It would mean, as is already stated in the first post, that the *default* curve would be no different normal stick movement, rather than a flat 50% as it is now on radio.

raphaelcoeffic, 2022-10-10:
Sort of: when creating a new curve on radio, it will default to `f(x)=x`, however, if you just select some unused curve in mixer, input or outputs, it will use `f(x)=0`.

gagarinlg, 2023-03-18:
Since the curve page now only displays configured curves, maybe the curve selection pup-up could do the same?

pfeerick, 2023-03-19:
You'll get complainers either way - those who select the curve as they setup the mixes, and then go setup up the curves afterwards, and then those who say it should only show configured curves. :shrug: It's no different with GVs... ie. use GV1, GV2 and then go setup the parameters for the GV afterwards. Or do you setup up GV1 and GV2, and then configure the mixes, LS etc?

PR #1078 (2021-11-20): Companion set new curves (-100,-100) to (100,100)
Fixes #1067 (on Companion side)

Summary of changes:
Assigning the proposed default to an input, mix or output has no effect on the target value and thus is safer especially if applied to a throttle.
Some housekeeping.