EdgeTX/edgetx #1069: Radiomaster Tx16s - Flashing receiver firmware via external module port gives a timeout

onki69, 2021-11-17:
I tried to flash a new firmware for my Radiomaster R168 using EdgeTX 2.5 and the external module port.
Everytime I tried it I got a timeout error.
When I used OTX 2.x with the same transmitter I could update the firmware without any problem. I addition it was possible with OTX using a Jumper T16.

pfeerick, 2021-11-18:
Please try a nightly build - this may have been resolved since 2.5 was released.

wimalopaan, 2021-11-18:
Well, don't think so, the R168 seem to be special. With actual main I can't update those either.

raphaelcoeffic, 2021-11-18:
> Well, don't think so, the R168 seem to be special. With actual main I can't update those either.

FrSky receiver are ok, but these are "difficult", as it seems. The issue is: I don't have any. RM told me they have been discontinued....

onki69, 2021-11-20:
Tried the Nightly (Oct 19th) with my RadioMaster and R168.
Still the same problem.
"Firmware update error"
"Device not responding"

I also tried updating with the same Nightly on my new Jumper T-Lite: No problem to flash a R168.
Seems to be a RadioMaster TX16s specific problem.

R168 is available at Hobbyking:
https://hobbyking.com/de_de/radiomaster-r168-d16-lbt-8ch-pwm-sbus-s-port-receiver-frsky-d16-compatible.html?queryID=3263bd5b7f0e28dbc379858f4ce39009&objectID=86447&indexName=hbk_live_products_analytics

I do have one R168 available. Depending on where you live I could send it over.

wimalopaan, 2021-11-20:
I have plenty of them, if soneone needs one, here is also an offer

wimalopaan, 2021-11-20:
I tried the same procedure with a Jumper T12 and R168 and EdgeTx (main-2.6.0). No luck either. Maybe be a rather difficult problem of HW and SW (EdgeTx) issue here

pfeerick, 2021-11-21:
lol... damn... I have R81/R86/R86C/R161 ... close, but not quite! I'll see if I'm lucky though and one of the others is also temperamental.

wimalopaan, 2021-11-21:
Ok, where to send?

stronnag, 2021-11-21:
My experience is that the device is extremely temperamental to flash.
* An older one from Banggood flashed without issue
* Newer models from a local supplier involved much frustration; short cable, twisted cable, untwisted cable, longer cable. Finally one cable option worked.
* There's a long thread of frustration and similar experiences on RCG.

pfeerick, 2021-11-21:
> Ok, where to send?

Either Raphael or Risto would be the ones to send to, not me ;)

raphaelcoeffic, 2021-11-21:
> Ok, where to send?

Just contact on Discord ;-)

rotorman, 2021-11-21:
@wimalopaan sends me one R168. I will do an analog trace of S.Port line during flashing under OpenTX and another under EdgeTX, then we will see the difference.

pfeerick, 2021-11-24:
> Ok, where to send?

Just for you, I ordered a pair of R168's when I found out earlier today Hobbyking still stock them (j/k... I wanted some anyway) ... but [reading the RCG thread](https://www.rcgroups.com/forums/showthread.php?3740151-Radiomaster-receivers) where people are talking about their woes trying to update them makes wonder WTH is going on with them... having to go to the trouble of making special short twisted cables in the hope that they would update... there must be something really wrong with that hardware!

wimalopaan, 2021-11-24:
I had not the time to dig into that, so I sent one to @rotorman . 
My guess is, that the pulldown for the serial line has wrong value (or missing). But I'm sure that @rotorman  will find out soon ;-)

dlktdr, 2021-11-24:
I have 5 of the R168's. Bought a few extra when I heard they were discontinued.. Never tried flashing one yet, all seemed to work great out of the box for me. I'll try programming a few OTX/ETX and send the scans to @rotorman he can compare with as well.

rotorman, 2021-11-24:
I have not yet received @wimalopaan 's R168's, after I do, my plan was to do an analog scan of the pins during flashing under OpenTX and under EdgeTX.

rotorman, 2021-11-24:
Received the R168 @wimalopaan sent me. Created the cable as of RM instructions https://radiomasterrc.freshdesk.com/support/solutions/articles/64000257762-r168-r161-receiver-firmware-update-procedures and downloaded the firmware for R168 https://www.radiomasterrc.com/articleb-139.html

What can I say, worked with EdgeTX and RM TX16S without any hiccups:
[image]

Hookup:
[image]

I should get tomorrow or on Friday the Logic Pro 8, I will nevertheless measure the signal with it as well.

@wimalopaan what problem did you exactly have - did the flashing not even start, or did it stop intermittently or was flashed corrupt, or something else?

wimalopaan, 2021-11-24:
The R168 I tried last time, I got "device not responding".
I will test one of those tomorrow myself once again.

wimalopaan, 2021-11-24:
mmh ... the next one works.
I think, I have to test them all...

[image]

rotorman, 2021-11-25:
Here an analog + digital signal trace in Saleae v2 Logic sw format ( https://www.saleae.com/downloads/ ) of power and S.PORT pin during a successful flashing of RM R168 with RM TX16S.
Not compatible with Saleae v1 Logic software unfortunately.

Due to GitHub single attachment size limit of 25MByte, I had to split the trace to 4 parts (remove the last *.zip in file names and unpack, e.g. with 7-zip):
[R168_TX16S_flash_successful.zip.001.zip](https://github.com/EdgeTX/edgetx/files/7604619/R168_TX16S_flash_successful.zip.001.zip)
[R168_TX16S_flash_successful.zip.002.zip](https://github.com/EdgeTX/edgetx/files/7604621/R168_TX16S_flash_successful.zip.002.zip)
[R168_TX16S_flash_successful.zip.003.zip](https://github.com/EdgeTX/edgetx/files/7604623/R168_TX16S_flash_successful.zip.003.zip)
[R168_TX16S_flash_successful.zip.004.zip](https://github.com/EdgeTX/edgetx/files/7604625/R168_TX16S_flash_successful.zip.004.zip)

wimalopaan, 2021-11-25:
Did you observe any anomalies?

rotorman, 2021-11-25:
Not with this combination of radio and receiver, no. All works as expected and also the signal looks fine. There is a tiny initial glitch on the S.PORT line, when the receiver (ext. module bay) gets powered, but that is all that was somehow out of perfect situation.

Upper two lines show the signals in digital, bottom two the same signals when sampled with analog capture simultaneously:

[image]

wimalopaan, 2021-11-25:
Looks really noise-free.
What is the upper voltage level of the S.Port Signal?

wimalopaan, 2021-11-25:
Unfortunately I burned my last un-flashable R168 a few weeks ago. All other R168s got flashed flawlessly.

I asked on used who has a un-flashable R168 to send me. Maybe we have luck.

rotorman, 2021-11-25:
> What is the upper voltage level of the S.Port Signal?

3.4 V
Here the stats over the whole flashing period of more than 1 minute:

[image]

raphaelcoeffic, 2021-11-25:
> > What is the upper voltage level of the S.Port Signal?
> 
> 3.4 V Here the stats over the whole flashing period of more than 1 minute:
> 
> [image]

It’s a nice new toy you have ;-) I hope mine is coming soon.

dlktdr, 2021-11-25:
I just tried with 3x R168's

2 - Flash perfectly. OTX2.3.14 + ETX 2.5
1 - Cannot program (ETX or OTX) - This one was new right out of the bag 👎 

## Good
[image]

## Faulty one
Looks like one the one I cannot program the s. port line is being held low.
[image]

Must be related to the R168 not the software. Or the shipped software on the R168 holding the pin low..

rotorman, 2021-12-08:
@wimalopaan sent me two confirmed problematic R168 units and I had a look at them. Both fail flashing with "Device not responding" error.

First problematic R168 mixed signal trace:
[image]

zoomed-in (both axes):
[image]

Second problematic R168 mixed signal trace:
[image]

zoomed-in:
[image]

The PPM signal does not go higher than 0.75 V with these problematic R168. I will open one up and look inside.

rotorman, 2021-12-08:
Well, this explains the low voltage - less than 9 Ohm between the signal pin and the GND:

[image]

dlktdr, 2021-12-08:
Same as the one I had too... Maybe your skills of following traces is better than mine. I traced it goes to this inverter but couldn't find where else and wasn't the inverter.  Thought about trying your method of sanding the board down ;)

It almost looks like some of the vias might be touching the ground plane but I didn't have a microscope good enough to see.

[image]

rotorman, 2021-12-08:
The signal wire is connected to the [74LVC1G04DBV](https://www.ti.com/lit/ds/symlink/sn74lvc1g04.pdf) input pin 2

[image]

on the part @dlktdr circled:

[image]

I do not see any series resistor not any pull-up/pull-down resistor. Possibly, due to flux not washed away, a higher voltage has creeped into the chip (as nothing is limiting it and CMOS input has very high input-impedance) and busted it. I will ask @wimalopaan is it is OK for him if I replace that chip for testing purposes on his R168 to find out.

rotorman, 2021-12-08:
Got OK from wimalopaan to mod the board, so de-soldered the inverter and at the same time washed the whole receiver PCB from factory flux with a special PCB cleaner.

[image]

Damn, still measure 8.5 Ohms between signal and GND. Need to further investigate, but cannot see any connection on that line on top and bottom layers, possibly the track is routed through internal layers.

dlktdr, 2021-12-08:
@rotorman.

If it's an Sport line,  I would fathom a guess that the other inverter right beside it is the output to the tristate c265. 
I never got around to trying but you might want to pop the c265 chip too to see if it's output is shorted?

rotorman, 2021-12-08:
I cannot see any shorts. Here some microscope shots of the signal line in question:

[image]

[image]

[image]

rotorman, 2021-12-08:
> If it's an Sport line, I would fathom a guess that the other inverter right beside it is the output to the tristate c265. I never got around to trying but you might want to pop the c265 chip too to see if it's output is shorted?

Good tip! Indeed I measure continuity (0.3 Ohm, so basically just track and connector resistance) to the following pin as well:

[image]

rotorman, 2021-12-08:
@dlktdr you were spot on. After desoldering the [74LVC1G126DBV](https://www.ti.com/lit/ds/symlink/sn74lvc1g126.pdf) and soldering back the [74LVC1G04DBV](https://www.ti.com/lit/ds/symlink/sn74lvc1g04.pdf) I now measure 1.5 MOhm.

[image]

[image]

rotorman, 2021-12-08:
I replaced the [74LVC1G126DBV](https://www.ti.com/lit/ds/symlink/sn74lvc1g126.pdf)

[image]

and we have success:

[image]

[image]

[image]

In order to get a replacement 74LVC1G126DBV, I temporarily sacrificed one of my TX16S mainboards... will order the part, next time I fetch components and resolder it back to the radio

dlktdr, 2021-12-08:
I was actually looking at your radio master schematics yesterday seeing how they did their sport for my own idea. 

On another note  I don't know if you noticed but when I was programming there is some pretty high transient spikes on the transitions on the s port line coming out of the radio master. 

I wonder if these might be damaging the chip? Not sure what the specs on that chip are. And Like you said, I don't see any protection at all on the lines.  I guess they discontinued it for a reason :)

rotorman, 2021-12-08:
I used Saleae Logic Pro 8 that is a mixed signal analyzer (analog + digital). I can see not higher levels than 3.4V during flashing (using 1.5 Msamples/sec):

[image]

wimalopaan, 2021-12-09:
Thank you for your work @rotorman !!!

wimalopaan, 2021-12-09:
I just got the next broken R168s, so I will do the same measurements on my own and then see, if they suffer from the same defect.
I only have about 10 pieces 1g126 in my smd-box. Maybe I should order some more 1g126 from digikey ...

pfeerick, 2022-06-01:
I think we can safely close this issue as the discussion has indicated that it an issue more specific to the R168 being either hard to flash (as RCG threads show OTX users have had just as many issues) or unit specific issues such as faulty inverters. I've also been able to update the 2 R168s I got from Hobbyking without issue - with both v1 and v2 ACCST firmware.

closed 2022-06-01 by pfeerick