EdgeTX/edgetx #1073: Move models.txt to \models

jmxp69, 2021-11-18:
If models.txt or models.yml can be moved to the models directory, it would be easier for people to move models from one tx to another by copying just the models directory. Given there is no radio specific data in models.txt(yml) it should be easy to implement this change.

pfeerick, 2021-11-18:
So this would simply be to make it easier to get a say a new transmitter up and running? As it's not much help with a transmitter that already has models on it. This would be the perfect time to do this change - as it could be `\RADIO\models.txt` => `\MODELS\models.yml`, therefore there won't be any confusion about where `models.txt` should be.

jmxp69, 2021-11-18:
Yes. There are a lot of people out there with more than one radio. This simply makes it easier to move/copy/backup models for people who chose to use a file manager or cli. There is no radio specific info in models.txt(yml) so why not store it with the actual model files?

pfeerick, 2021-11-18:
Well, it does, and it doesn't help... i.e. if I copy from my T-Lite to my TX12 (hardware switch variations aside), I would have to rename models that conflict so that I don't blow away my existing models, and then manually edit models.yml to add the new ones. 

I fully agree with this as part of a 'clean slate' copy, but for anything more, I don't think it is the right approach. Perhaps a second folder that you copy new models to, and an import facility on the radio UI? (or even something a bit like the backup and restore that was present on the B&W radios) Or is this going too far into the territory of, um, so why do we need Companion?

PR #1128 (2021-11-22): Move YaML models list to /MODELS
Resolves #1073
commit 5290820a (2021-11-22): Move YaML models list to /MODELS

Resolves #1073
closed 2021-11-22 by raphaelcoeffic