EdgeTX/edgetx #1077: After the last change to libopenui scroll "inertia" feature is lost

eshifri, 2021-11-18:
**Describe the bug**
When scrolling the window, the motion stops immediately when the finger motion is stopped.

**To Reproduce**
Open Model Setup window and scroll to the bottom.
The scrolling stops when the finger is removed from the screen.

**Expected behavior**
"Inertia" was a very nice feature.

**General information**
 - EdgeTX current code
 - Hardware  RM TX16S

gagarinlg, 2021-11-20:
this should be fixed with [https://github.com/EdgeTX/libopenui/pull/37](url)

closed 2021-11-21 by pfeerick

pfeerick, 2021-11-21:
I'm not sure yet this is 100% resolved, as inertia does seem to be back, it is still a bit *latchy*, but we'll see what others think at the next nightly...

eshifri, 2021-11-21:
I think it used to be more convenient.