EdgeTX/edgetx #1083: RM TX16s Initial convert to yaml LS1 and LS2Calibration missing

NemesisXB, 2021-11-19:
**Describe the bug**
After updating to a version that uses yaml, LS1 and LS2 calibration was invalid.

**To Reproduce**
Start with binary model and radio files, have the RM TX16s do the SD Card conversion on startup. Check the LS1 and LS2 data in analog screen.

**Expected behavior**
LS1 and LS2 should be calibrated as they were before the SD Card upgrade.

After redoing the calibration and comparing the original converted yaml file to the yaml file after recalibration, LS1 and LS2 is clearly missing from the yaml file (on the left SD Card converted version, on the right, after recalibration):
[image]

**General information**
 - EdgeTX #8f55fb0f
 - RM TX16s

onki69, 2021-11-21:
I have a similar problem. I can't even calibrate properly.
The lower part of the throw (-100 to 0) is completely missing.
Is this caused by this bug?
I can't find a way to get LS and RS back to full throw.

NemesisXB, 2021-11-21:
> I have a similar problem. I can't even calibrate properly. The lower part of the throw (-100 to 0) is completely missing. Is this caused by this bug? I can't find a way to get LS and RS back to full throw.

Sounds like a different problem. Mine calibrated without issue.

commit 06c04c85 (2021-11-26): Use symbolic indexes for calibration data

Resolves #1083
commit a3f8dc05 (2021-11-27): Use symbolic indexes for calibration data

Resolves #1083
closed 2021-11-26 by raphaelcoeffic