EdgeTX/edgetx #1084: Voice prompt on LS slider after upgrade to yaml

NemesisXB, 2021-11-19:
**Describe the bug**
Upgraded to version that uses yaml. For some unexplained reason I have a voice prompt saying "Timer 1 up" every time I pass the center point of the LS slider. No matter the direction, sliding it up passed the midpoint and sliding it down past the midpoint both trigger the voice prompt.

**To Reproduce**
Attached radio.yaml and model.yaml files.

**Expected behavior**
This prompt was not there before the SD Card upgrade to yaml.

**General information**
 - EdgeTX #8f55fb0f
 - RM TX16s

NemesisXB, 2021-11-19:
OK, This seems to be a duplicate of #1062. So LS and RS were triggering the TIMER 1 and TIMER 2. But why the voice prompt even though I don't have a prompt configured for it?

NemesisXB, 2021-11-19:
OK, Flashed #1080, but don't think this solved it completely. It went from saying "Timer 1 up" when crossing the center point to haptic feedback and high pitched "ting" sound.

eshifri, 2021-11-19:
> OK, Flashed #1080, but don't think this solved it completely. It went from saying "Timer 1 up" when crossing the center point to haptic feedback and high pitched "ting" sound.

This is the expected behavior, isn't it?

NemesisXB, 2021-11-19:
> > OK, Flashed #1080, but don't think this solved it completely. It went from saying "Timer 1 up" when crossing the center point to haptic feedback and high pitched "ting" sound.
> 
> This is the expected behavior, isn't it?

It never did it before. And also only does it for LS and not RS. So no, not expected behaviour.

eshifri, 2021-11-19:
You have selected these actions by pressing L virtual button.
There is another bug #411 : the L and R buttons are not highlighted.
You can control beep and haptic by pressing these buttons.
If you leave model setup screen and return back the button(s) will be highlighted properly.

NemesisXB, 2021-11-19:
> You have selected these actions by pressing L virtual button. There is another bug #411 : the L and R buttons are not highlighted. You can control beep and haptic by pressing these buttons. If you leave model setup screen and return back the button(s) will be highlighted properly.

Doh, that was it. However mine never shows L. Just a blank block.

closed 2021-11-19 by NemesisXB

pfeerick, 2021-11-20:
> However mine never shows L. Just a blank block.

This is what #1080 should have solved ;) Then you run afoul of #411 if you had centre beeps enabled (as you can't see it on the TX).