EdgeTX/edgetx #1086: Dynamic number of an items (GV, LS, CURVES etc.)

NemesisXB, 2021-11-19:
**Is your feature request related to a problem? Please describe.**
It was mentioned that once yaml is implemented that stuff like like global variables does not have to be capped at the current limit of 9. And it can be created dynamically as the user requires them. I think this can be done for a lot more than just the global variables.

**Describe the solution you'd like**
At the very least this can be done for the following:

- Global variables
- Logical Switches
- Special Functions
- Curves

It would be great if it could also be extended to the timers, currently there are 3 timers, I use none of them, but what if somebody wants 4.

And it would also be awesome if it could be extended to flight modes as well. I hate seeing the 8 flight modes as I am not using it. Almost as much as seeing the heli tab. And if it is dynamic, someone that might want more than 8, can do it.

I know that this will at the earliest probably only be done for version 2.7. I am just creating a ticket so it is here.

gagarinlg, 2021-12-14:
We can just show the list until and including the last used entry plus one. When you then use the last row, an additional row will be shown. It would be possible to generate unused rows in the middle of the list, but that can't be helped so easily. 

Having more timers is a different task, though.

raphaelcoeffic, 2021-12-14:
One thing is the UI, which is just a matter of coding it. The storage model has been prepared for more flexibility as well, and we will see how we can push the boundaries in the next releases. So we should see these features coming, one by one, as they get implemented.

gagarinlg, 2021-12-14:
For the GVAR list and similar stuff, there is no need to change the storage, just how it is displayed.
Timers are, of course, a different thing.

fhpage, 2023-09-06:
Increasing the number of GVAR's (as well as Logical Switches, Special functions) would be game changer for complicated programs. Will this ever happen?

raphaelcoeffic, 2023-09-07:
@fhpage it will, but you will have to be patient. Depending on the item, it has some implications on the internal memory model that make it quite complex to integrate. However, we need this to reduce the memory usage for most people who don’t use that many items, while allowing for more complex models in some cases.