EdgeTX/edgetx #1088: Split window touch top carousel issue

NemesisXB, 2021-11-19:
**Describe the bug**
Referring to Issue #981 merged in #1002. First off, I think this is not a good way of achieving it. It is **unintuitive**, and I can't think of any other established UI design that implements anything similar. As an analogy, a dropdown box (when not selected yet) always starts at the top, not randomly based on where you clicked on it. I know it is majority rules, but personally think this should be removed.

Now on to the actual issue:
I have 2 main views, Main view 1 and main view 2. When main view 1 is shown, the split top touch works as intended. However, when I go to main view 2, no matter where I click in the top bar, it always open the second half of the menu.

**To Reproduce**
Create a home screen with 2 main views, swipe to the second main view and tap anywhere in the top bar, the second part of the menu is always opened.

**Expected behavior**
How I expect it to work is always open the menu in the same way. So I expect this to be removed. Going forward the top bar widgets could be shortcuts to different items, like clicking the date and time takes you directly to where you can set the date and time. With this change kept, this will not be possible.

However should removing this change not be the decision, At least have it consistent between multiple views.

**General information**
 - EdgeTX #1002fddd
 - RM TX16s

pfeerick, 2021-11-20:
For the moment, lets just fix this buggy behaviour, and then revisit the UI behaviour generally. I'm starting to agree with this, although as not being 'user friendly' whilst I also like that this lets you access the second half of the menu without having to swipe or touch the roller - i.e. to access reset telemetry.

NemesisXB, 2021-11-20:
> For the moment, lets just fix this buggy behaviour, and then revisit the UI behaviour generally. I'm starting to agree with this, although as not being 'user friendly' whilst I also like that this [...]

I'm happy with this. I think a better way of solving this is to allow for the customisation of that menu. Have a user hide or show certain items, and reorder the menu. Then you can have the items you use most often always on the first page. There **seems** to be enough space to show 5 items as well in stead of just 4. At least on landscape displays.

pfeerick, 2021-11-20:
I think this would be a future goal indeed. Or even making the menu multi-line... as it could quite easily be two lines on TX16S style  display, and probably 4 lines on NV14 style display. But that will be a 2.7 discussion I think ;)

PR #1090 (2021-11-20): fix split touch window for carousel #1088
Fixes #1088 

Summary of changes:
take into account that for multiple main screens the x coordinate may be bigger than the screen width.
closed 2021-11-20 by pfeerick