EdgeTX/edgetx #1091: Widget is stuck in full screen

oleg-sherman, 2021-11-19:
*Before submitting,  search the open issue if there is already a report of the bug*

Also often minor bugs are already fixed in nightly builds, often it is
a good idea to test (in a controlled environment like a bench setup)
with a nightly firmware to see if the bug still exists.

**Describe the bug**
When users enters full screen for a single value widget, it is stuck in full screen, and the only way to go back is to turn off/on the radio.

**To Reproduce**
1. Set single value telemetry
2. Open it in full screen
3. Can't get back from full screen

**Expected behavior**
The user should be able to go back from full screen

**Workaround**
Turn off and on the radio

**General information**
 - EdgeTX version 2.5.0
 - Hardware TX16s

NemesisXB, 2021-11-19:
Have you tried long pressing the RTN button?

oleg-sherman, 2021-11-20:
Yep, that solves it, IMO long press is less intuitive, maybe changing it to regular press will be better...

closed 2021-11-20 by oleg-sherman

pfeerick, 2021-11-20:
Long RTN is used to exit both standalone scripts and fullscreen widgets, leaving a short RTN press for use within the app.