EdgeTX/edgetx #1092: implement working " New Model Wizard "similar to OTX 2.3

Airfiddler, 2021-11-20:
*Before suggesting a new feature here, please search the open issue if there is already a request for the feature.
You can add your comment there, or react so that we know know much interest there is in the feature request.*

**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

Airfiddler, 2021-11-20:
This would reduce the learning curve for new ETX users and get them flying faster.

raphaelcoeffic, 2021-11-21:
I think there are two things here:
- There is nobody at the moment to maintain the Wizards
- We thought about having them in the "tools", as most people are annoyed when the Wizard is starting on every new model.
- Wizards are dull. The idea we had was to have different modes: "Starter", "Advanced" and "Expert". The starter would obviously have the Wizard starting and a simple menu without too many options. "Experts" would be like now with all the bells and whistles.

Airfiddler, 2021-11-22:
Good points. I would think the majority of ETX users would not fall into
the beginner category. I believe the category idea has tremendous merit, as
there would be something for everybody in ETX. The lifeblood of any pursuit
lies in attracting new users. The catagory option should probabily should
allow the noobs to turn off the feature as they progress into the advanced
catagory. Kudos to the developers on their amazing progress so far. Keep up
the great work.

gagarinlg, 2021-11-26:
is this a duplicate of #8 ?

pfeerick, 2021-11-26:
Yeah, I'll go with that... 

Duplicate of #8

closed 2021-11-26 by pfeerick