EdgeTX/edgetx #1097: NV14 RF Signal (telemetry)

nightflasher, 2021-11-20:
have searched but not found.

When i bind my Fli14+ rx to my NV14 (current nightly build), the NV14 repeatedly "rf signal critical".
Also the db measurement shows 20db lower then the last openTX build by Stefan and Jacob (version from fpv-racing-forum).

in openTX it shows about 73db and edgeTX whows only 53db at the same distances.

pfeerick, 2021-11-20:
@eshifri You played with the RSSI and warnings, didn't you? Any thoughts?

eshifri, 2021-11-20:
Yes. 
The RSSI signal now is the native FlySky value; please notice that it is negative and the range os [`~ -90db, -60db].
(This was optional in FlySky OTX code: there was a checkbox to switch to this code.)
Warning is working on Sgnl now. This was the goal in OTX version that was made possible after th eradio module FW update, but was not implemented in the Jacub's code. 
"Sgnl" has the natural range [0, 100]. You should configure the alarms accordingly.
There is an annoyance of getting "signal too low" once, when the link with  RX is established. I do not know how to fix this one in a proper way.

pfeerick, 2021-11-21:
I don't know what triggers that in this instance, so this may be completely unrelated, but ExpressLRS v1 (fixed in v2) radio link had an issue where it was the receiver's fault, as the very first value that was sent was low - possibly a zero - so turn on TX, turn on quad/RX, "signal critical" or whatever, and then it was fine.

eshifri, 2021-11-21:
It was not happening in Jacub's code, so...

CharlesGauci, 2021-12-06:
Hi Eshifri, I have been having the 'rf signal critical' warning every time I switch the receiver on since Jacub's last update. I mentioned it in RC Groups 'Underground FPV Dark Knight Radio' on 6.12.2020 post number 3883 but Jacub did not reply.

eshifri, 2021-12-06:
Hi @CharlesGauci Can you please clarify: do you get this warning warning one - when link is established, or it happens all the time?
Also please notice that levels for these warning are configurable (on the telemetry setup page).
EdgeTX and NV14 OTX are using different parameters for connection quality: OTX was using rescaled RSSI; ETX - Sgnl (signal quality).
So you may nedd to do some changes for your setup.

CharlesGauci, 2021-12-07:
Hi Eshifri, I only get the warning once, as soon as I switch the receiver on. I have not converted to EdgeTX yet as I do not have the time to rebuild the model setups of all my fourteen model airplanes, but I will do it on a new memory card in the near future. Meanwhile I will try to find the configuration settings on the telemetry setup page as you suggested. Thanks.

eshifri, 2021-12-07:
Aha, I also have the same problem in EdgeTX. I think it was not happening in OTX, but I'm not sure. 
I consider it a minor nuisance. No idea so far how to fix it.

uPz3TT3r, 2021-12-11:
I have the exact same problem as the thread starter. My Flysky Receivers all cause critical signal voice Output all the time. Changing Offset of sgnl or something to have sgnl 90 does not stop voice Output.

uPz3TT3r, 2021-12-11:
[image]

thats the Screen with Quad very close. I had to turn off telemetrie to Not hear the sound anymore

eshifri, 2021-12-12:
@uPz3TT3r Can you please check the version of your radio module FW?
Looks like your TX is rather old and have not upgraded the radio module FM: in the current versions the signal is suppoesd to be in the [0, 100]  range. In the original version it used to be [0, 10].
Unfortunately, ETX does not provide the tool to update the radio FW - OTX (Jacub's version) - does.
What happens here is the following: for old FW ETX uses RSSI for warning, but as you can see RSSI is negative and the alarm levels cannot be set to negative numbers.

uPz3TT3r, 2021-12-12:
[image]
 Do you need this Info? I dont know

eshifri, 2021-12-12:
Yes, that is what I was asking about. 
Unfortunately, I have no explanation now: with this FW Sgnl should be 100 - not 10.

uPz3TT3r, 2021-12-12:
Cloud it be receiver related? I have the modded rssi Firmware (rssi on aux 10) on all receivers

eshifri, 2021-12-12:
I do not know for sure, but I would be surprised. AFAIK all the changes were in the radio module FW.
You are getting excellent RSSI: -60db, so Sgnl should be also the highest possible. And 10 used to be the highest in older radio module FW versions.
As a work-around you can disable warnings on the telemetry page and create your own ones using special functions. (I would say RSSI== -78 is a warning and -82 is critical.)

Unfortunately, I do not see what can be used as a trigger for special processing, since your RM FM version looks new.

uPz3TT3r, 2021-12-12:
I never had this Problem with the otx Versions for Nirvana. Is it not possible to handle it the same way?

eshifri, 2021-12-12:
In NV14 OTX RSSI was re-mapped into something resembling [0, 100] range. It was a non-linear signal without clear physical meaning: neither percentage of packets received (Sgnl), nor RSSI (in dB). RSSI numbers displayed were not RSSI provided by FlySky receiver and used by every other FlySky system.
There was a check-box on the telemetry screen  allowing to switch to "native" RSSI (I used it for all my models).
In ETX this check-box is gone, but instead we are using FW version: if FW generates Sgnl in [0, 100] range with high resolution we use it instead of RSSI. 
For old FW we perform the same re-mapping procedure, since people are used to it.
The case you are reporting is something I do not understand: you have the new FW.
I will try to figure out how this is possible and find the different criteria for making the switch, but no promises at this point.

If you still have OTX around you may want to try to re-flash the RM FW using this one: 
https://www.rcgroups.com/forums/showpost.php?p=45357535

Aaronsss, 2021-12-31:
I am experiencing the same issue with 2.6.0-RC1 on my nirvana with an FLI14+

The radio constantly shouts out RSSI critical and its not possible to reduce the RSSI warning so this doesn't happen. In addition, the signal indicator in the top bar is always empty. These issues did not happen in the latest flysky version

I have disabled the warnings for now but would be awesome if there were a better fix

eshifri, 2021-12-31:
I would strongly suggest to update the radio module FW to the latest version from qba667  (see link above).
It has some additional advantages except fixing the RSSI warning.
I do not know how it happened that two different FWs have the same version. Due to this there is no reliable version to fix this. At least I do not see any at the moment.

eshifri, 2022-01-01:
Can you please give a try to this PR: https://github.com/EdgeTX/edgetx/pull/1329 ?

Aaronsss, 2022-01-01:
How do I try it? Where can I find the bin file?

Also with the radio module FW update I assume I do that by going back to the old flysky version, put the update file in the firmware folder and then update it from within the radio's file browser?

eshifri, 2022-01-01:
You can find the new FW here: https://github.com/EdgeTX/edgetx/suites/4803597875/artifacts/134853144

Yes, in order to update the radio module FW you will have to go back to OTX and use SD card browser to select RM FW.

Aaronsss, 2022-01-01:
Yep no more warnings :). The only thing I would say is it may be worth tweaking the scaling for the rssi meter on the top bar as i can't get that last bar to light up

[image]

eshifri, 2022-01-01:
Done.
For old RM FW the RSSI range will be [40, 100].

PR #1329 (2022-01-08): Improve support for old NV14 Radio Module FW
Fixes #1097 

Summary of changes: if the RM FW version is old and the Sgnl value is low [0, 10] use scaled RSSI (OTX formula) for alarm.
closed 2022-01-08 by pfeerick