EdgeTX/edgetx #1103: NV14 widgets

nightflasher, 2021-11-20:
I forgot to mention the widget problem I had! 
I like to see three things on my first screen:
- RSSI (works) 
- model name and picture (picture not tested) 
- output (nothing shown, full screen worked)

pfeerick, 2021-11-21:
Multiple smaller easily digestible issues are good ;)

For NV14, output widget may need a full screen width layout.

Rip-Hunter, 2022-01-08:
Why can't you make outputs in the quarter of the window and what would be displayed how many channels, how many fit into the size of this small window? It's more convenient than having a separate screen with outputs.

Rip-Hunter, 2022-01-08:
And the same question: how can I close, for example, outputs, if I expanded them to the whole screen by double-clicking?

pfeerick, 2022-01-08:
Closing full-screen widgets is the same on all transmittes... long press RTN. In the case of the NV14, RTN is emulated by pressing the left trim joystick down. Something will need to be considered for other touch-screen only transmitters, so there could be a reserved space of onscreen control or something in the future.

closed 2022-01-17 by pfeerick