EdgeTX/edgetx #1109: Jumper T-Lite with CC2500 - Binding beep does not stop (Nightly Oct 19th)

onki69, 2021-11-20:
I have a small problem with my Jumper T-Lite and binding.
When I start the binding  prcedure an interval beep is indication the binding function.
When I stop binding by pressing "Ent" the Binding is no longer active (Binding is not blinking) but I still hear the beeping.
Only a power cycle will stop the beep.

If I remember correctrly the Binding was only active for a limited time and there was no need to stop it manually. Was there a change?

Bug or feature?
Regards
Onki

pfeerick, 2021-11-21:
Binding is active for a limited time for some protocols, but if you press the bind option a second time, it should also cancel bind and stop beeping. If it doesn't, it a bug ;) I'll check on my TX12S tomorrow, as it has CC2500 also (probably won't be able to repo on T-Lite as mine has the full 4in1). What protocol - frsky?

TNorthover, 2022-01-08:
I'm seeing the same thing on tx16s, Futuba protocol. Never looked at the code before so I may be wildly off base, but it seems like for Multi-protocol modules the beep is on whenever `getMultiBindStatus` is anything except `MULTI_BIND_NONE` (if I'm reading `pulses.h` correctly).

There's code in the older GUIs (I checked `128x64/model_setup.cpp`) to set it back at the end of a bind operation, but not in `colorlcd/model_setup.cpp` as far as I can see.

pfeerick, 2022-01-08:
Oh, interesting... so maybe bind end isn't being signaled or something for some protocols... Can repo that one... on TX16S and T-Lite... thank you! :)

raphaelcoeffic, 2022-10-16:
We've fixed this already a couple times in different other issues. Let's close this one for now.

closed 2022-10-16 by raphaelcoeffic

pfeerick, 2022-10-17:
This reads to be a duplicate of the recent report in #2263. https://github.com/EdgeTX/edgetx/pull/2455 should have resolved this behaviour for both colorlcd and B&W ;)