EdgeTX/edgetx #1110: Second condition of the Edge LS is not saved in YAML

eshifri, 2021-11-20:
I remember it was discussed in discord, but I do not see it logged here.

**Describe the bug**
tHE FOLLOWING IS THE trace OUTPUT FROM SIMULATOR:
logicalSw: 
   0:
      func: FUNC_EDGE
      def: **SA0, 3,**
      andsw: NONE
      delay: 0
      duration: 0

**To Reproduce**
Create Edge LS set both conditions and switch the radio off.
Restart the radio and look at the LS: the second condition will be set to either "--" or "<<".

**Expected behavior**
Condition should be saved in the nodel.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**General information**
 - EdgeTX version current
 - Hardware TX16S, NV14, simulator.

eshifri, 2021-11-20:
static const struct YamlNode struct_LogicalSwitchData[] = {
  YAML_IDX,
  YAML_ENUM("func", 8, enum_LogicalSwitchesFunctions),
  _**YAML_CUSTOM("def",r_logicSw,w_logicSw),**_
  YAML_PADDING( 10 ),
  YAML_PADDING( 10 ),
  YAML_SIGNED_CUST( "andsw", 9, r_swtchSrc, w_swtchSrc ),
  YAML_PADDING( 1 ),
  YAML_PADDING( 2 ),
  YAML_PADDING( 16 ),
  YAML_UNSIGNED( "delay", 8 ),
  YAML_UNSIGNED( "duration", 8 ),
  YAML_END
};

Looks like there is just no place-holder for additional parameter.

raphaelcoeffic, 2021-11-20:
@eshifri serialization / deserialization is all done in these 2 custom functions which can be coded freely.

eshifri, 2021-11-20:
@raphaelcoeffic 
My bad. Now I do not see what the problem is, but it definitely exists.

pfeerick, 2021-11-21:
Perfect... so that should solve this also then? https://github.com/EdgeTX/edgetx/issues/1048

eshifri, 2021-11-21:
I'm not sure, since I have not seen it happening.
This might be conversion issue. If re-saved in YML it shows "--"

pfeerick, 2021-11-21:
I'm pretty sure this is the same issue ... you're talking about the edge LS and V2 parameter going stupid... i.e.  "--" or "<<" ;) ... it's not conversion related - a restart of the radio running a nightly will go from the value you set to `<<` ;) That's why I'm hoping the PR will fix both issues together ;)

PR #1113 (2021-11-21): Save third value for EDGE LS in YAML
Fixes #1110 

Summary of changes: v3 for EDGE LS was not saved in the model.yml file.
closed 2021-11-21 by raphaelcoeffic