EdgeTX/edgetx #1116: Generated model YAML contains NUL byte

stronnag, 2021-11-21:
Edge main branch of 2021-11-21 (620a43a9c3bc).
Attempted to test loading model YAML into ruby (and python scripts). Both complained of illegal character.  The python example even told me where it was.

In the `logicalSw:` section, there appears an illegal NUL (\000) byte. Indicted as `\0` below in the T12 string in the `2:` element. 
```
logicalSw:
   0:
      func: FUNC_VNEG
      def: tele(11),112
      andsw: !L2
      delay: 50
      duration: 0
   1:
      func: FUNC_VNEG
      def: tele(11),109
      andsw: NONE
      delay: 50
      duration: 0
   2:
      func: FUNC_STICKY
      def: T\012,NONE
      andsw: SF2
      delay: 0
      duration: 0
```

raphaelcoeffic, 2021-11-21:
I believe this is even actually the octal character `012`. Have you checked with a hex editor the raw payload?

stronnag, 2021-11-21:
> I believe this is even actually the octal character `012`. Have you checked with a hex editor the raw payload?

The yaml file contains a NUL byte. The python error is
```
    raise ReaderError(self.name, position, ord(character),
yaml.reader.ReaderError: unacceptable character #x0000: special characters are not allowed
  in "model1.yml", position 4043
```
The `\0` is an artefact of pasting from emacs.
Original report edited for clarity.

stronnag, 2021-11-21:
Out of interest, about 50% of my saved models show this (12 are OK, 13 fail with a 'binary' character) in the YAML.

stronnag, 2021-11-21:
Original LS setting:
[image]

raphaelcoeffic, 2021-11-21:
@stronnag Thx for the info! Can you please verify it is fixed by #1118 ?

stronnag, 2021-11-21:
> @stronnag Thx for the info! Can you please verify it is fixed by #1118 ?

Sure, how do I get the radio to regenerate the YAML?

raphaelcoeffic, 2021-11-21:
> > @stronnag Thx for the info! Can you please verify it is fixed by #1118 ?
> 
> Sure, how do I get the radio to regenerate the YAML?

Just delete `/RADIO/radio.yml`, and it will re-generate from the `.bin` files.

stronnag, 2021-11-21:
Thanks, it' better: My test script now gives OK 19, failed 6 (vice 13/12 prior to #1118).
I need to look in detail at the fails and I'll update here. Probably tomorrow now.

raphaelcoeffic, 2021-11-21:
> Thanks, it' better: My test script now gives OK 19, failed 6 (vice 13/12 prior to #1118).
> I need to look in detail at the fails and I'll update here. Probably tomorrow now.

Cool! Probably another field now. But that seems to be a good test!

stronnag, 2021-11-21:
> > Thanks, it' better: My test script now gives OK 19, failed 6 (vice 13/12 prior to #1118).
> > I need to look in detail at the fails and I'll update here. Probably tomorrow now.
> 
> Cool! Probably another field now. But that seems to be a good test!

It's a different conformance problem. 
There are no non-conforming bytes in any of my samples **so this PR is fixed**.
I'll look at the remaining non-conformance issue tomorrow; I appreciate that's it's an optimised specific implementation, so it will be a different issue for you to decide if it's something you even want to fix.

PR #1118 (2021-11-21): Fix length of Txx string
Resolves #1116
commit 3ffa531f (2021-11-21): Fix length of Txx string

Resolves #1116
closed 2021-11-21 by raphaelcoeffic