EdgeTX/edgetx #1119: "Hello"-Track played last in row on startup...

notyourbiz, 2021-11-21:
On Startup first all the Tracks for Switch-state where played, "Hello"-Track is the last.

Steps to reproduce the behavior: Switching on and off

**Expected behavior**
"Hello"-Track first in Row

**Screenshots**
N/A

**General information**
 - EdgeTX version 2.5 / 2.6
 - Hardware TX16S

raphaelcoeffic, 2021-11-21:
@notyourbiz I think we have this issue already a couple of times :-D It's probably the first thing everyone notices 😅

rotorman, 2021-11-21:
@notyourbiz Thank you for taking the time to file the issue! In this case, the startup sound issue has been already listed in Issue #135. As this is a duplicate of the same issue, I will close it here. Cheers!

closed 2021-11-21 by rotorman

pfeerick, 2021-11-22:
Duplicate of #135