EdgeTX/edgetx #1121: Swith-Sate Tracks on Startup

notyourbiz, 2021-11-21:
**Is your feature request related to a problem? Please describe.**
Just my Ears... ;-)

**Describe the solution you'd like**
Selecable if Switch_State Tracks should be played on Startup or not

raphaelcoeffic, 2021-11-21:
What do you mean actually with switch-state tracks? Are these the “play” functions set up in “special functions”?

notyourbiz, 2021-11-21:
exactly...

could be a little annoying if the radio tells u every startup the actual state of every switch... i love to hear a track, if i change a switch-state, but not on startup for all switches...

raphaelcoeffic, 2021-11-21:
Some do like it to know the state of the model switches before they turn it on. It’s probably a matter of taste.

notyourbiz, 2021-11-21:
of course... thats why i suggested selectable...

somewhere in the Radio-Settings a checkbox [yes] [no] [maybe] ;-)

so anybody can choose...

raphaelcoeffic, 2021-11-21:
> of course... thats why i suggested selectable...
> 
> somewhere in the Radio-Settings a checkbox [yes] [no] [maybe] ;-)
> 
> so anybody can choose...

Curious to know what you figured should happen with “maybe” 😅

notyourbiz, 2021-11-21:
[maybe] is a future feature, like release 3.x, when its "really" talking to you...

Radio on startup: "hey, wanna hear the switches?"
You: "shut up, get me some coffee and start the planes engine..."

or so... :-)

pfeerick, 2021-11-22:
There is a ! condition in the play track special function you can use, so that switch state is *not* played at startup. i.e. I have audio set up on four switches I regularly use (i.e. arming, modes switch 1, modes switch 2, gps modes, flip recovery, etc), and only some of those will talk at startup.

notyourbiz, 2021-11-22:
@pfeerick: you really made my day!!! Awsome, thats what i was looking for... even better than my suggestion...

THX

pfeerick, 2021-11-22:
No problem! Glad to help :)

closed 2021-11-22 by pfeerick