EdgeTX/edgetx #1122: Generated YAML has blank field, failing validation.

stronnag, 2021-11-21:
Current master 2021-11-21, following on from #1116.

Generated YAML has blank field in `csutomFn` block, causing parser failure with ruby's YAML parser, element `9:` in sample:
```
customFn: 
   0:
      swtch: !SF0
      func: PLAY_TRACK
      def: armed,1x
   1:
      swtch: !SF2
      func: PLAY_TRACK
      def: darmed,!1x
   2:
      swtch: SF0
      func: RESET
      def: Tmr2,1
   3:
      swtch: SF0
      func: RESET
      def: Tmr1,1
   4:
      swtch: SA0
      func: PLAY_TRACK
      def: hirat,1x
   5:
      swtch: SA1
      func: PLAY_TRACK
      def: midrat,1x
   6:
      swtch: SA2
      func: PLAY_TRACK
      def: lowrat,1x
   7:
      swtch: L3
      func: PLAY_VALUE
      def: tele(2),30
   8:
      swtch: L2
      func: PLAY_TRACK
      def: batlow,!1x
   9:
      swtch: L4
      func: INSTANT_TRIM
      def: ,1
   10:
      swtch: L4
      func: FIRST_WITHOUT_ENABLE
      def: Sirn,1x
```
 
[image]

stronnag, 2021-11-21:
FWIW,
```
./ytest.rb models-yml/model10.yml
/usr/lib/ruby/3.0.0/psych.rb:456:in `parse': (models-yml/model10.yml): did not find expected node content while parsing a block node at line 313 column 12 (Psych::SyntaxError)

```

raphaelcoeffic, 2021-11-22:
@stronnag PR #1126 should fix this. However, I did not test anything, apart from running the unit tests which do not use the `INSTANT_TRIM` function. Any function without any parameter apart from "active" should behave the same.

stronnag, 2021-11-22:
Confirmed fixed by #1126. All my (25) models are now readable with the Ruby YAML module without error.

PR #1126 (2021-11-22): Fix attribute comma separation in custom functions
Resolves #1122
commit fad3ce67 (2021-11-22): Fix attribute comma separation in custom functions

Resolves #1122
closed 2021-11-22 by stronnag