EdgeTX/edgetx #1124: POTS and SLIDERS warning works incorrectly

eshifri, 2021-11-21:
**Describe the bug**
On the model setup screen Pot positions and & Slider positions are incorrect:
1. Pot positions are displayed independently of their availability: EX1 - EX4 are always displayed. This is easy to fix.
2. RS cannot be enabled. This must be due to the fact that g_model.potsWarnEnabled is uint8_t and cannot held all potentiometers for X10 (TX16) targets.
Changing this field to uint16_t causes static_assert() and I'm not clear whether it has rippling effect across the whole system.
Advise will be appreciated.

**To Reproduce**
Open Model Setup Screen.
Scroll down to Pots and Sliders.
Observe S1 - EX4  pots being shown.
Click on RS - it will not be highlighted.
Settings for LS and RS are not applied when the TX is switched on.

**Expected behavior**
Should work :-)

**General information**
 - EdgeTX current
 - Hardware  RM TX12S, simulator

eshifri, 2021-11-22:
So I have figured out that it is possible to change the type from uint8_t to uint16_t, but it requires changes in many places.
All yaml_datastructs_*.cpp are to be changed. Is there any reason to use the hard-coded number of in bits rather than sizeof(struct.field)*CHR_BIT?

It should be also noted that TR_VSRCRAW contains more than 8 fields for many targets. Looks like it was just a matter of luck that nobody ran into this problem before.

raphaelcoeffic, 2021-11-22:
> So I have figured out that it is possible to change the type from uint8_t to uint16_t, but it requires changes in many places. All yaml_datastructs_*.cpp are to be changed. Is there any reason to [...]

They are actually generated. So it’s just a matter of changing what needs to in the actual structures you need, and re-generating.

eshifri, 2021-11-22:
How can I do that?
Does it happen when the cmake is called?

raphaelcoeffic, 2021-11-22:
@eshifri you should use the docker container together with this script: https://github.com/EdgeTX/edgetx/blob/main/tools/generate-yaml.sh

PR #1129 (2021-12-07): pots warning enabled fix
Fixes #1124 

Summary of changes:
1. Do not display non-existent potentiometers in the warnings group.
2. Increase the size of the potsWarnEnabled field in the ModelData to 16 bits in order to save all possible potentiometers and sliders.
closed 2021-12-07 by pfeerick