EdgeTX/edgetx #1131: [YAML] Problem with !switch elements

stronnag, 2021-11-22:
My aim is to be able to "round trip" YAML files though ruby or python to make systematic cosmetic changes (e.g. globally change widget text colour).

The following construct causes problem:
```
logicalSw:
   0:
      func: FUNC_VNEG
      def: tele(6),1120
      andsw: !L2
      delay: 50
      duration: 0
```
* The standard python YAML parser fails:
```
could not determine a constructor for the tag '!L2'
  in "models-yml/model24.yml", line 434, column 14
```
* The file fails to round trip correctly via the ruby YAML parser. I suspect this is a silent failure in the parser, the round-trip output (YAML => ruby object => YAML) is missing the `andsw:` value:
```
logicalSw:
  0:
    func: FUNC_VNEG
    def: tele(6),1120
    andsw:
    delay: 50
    duration: 0
```
As suspected, the ruby parser dislikes / silently ignores the '!L2' value, ruby object:
```
"logicalSw"=>
  {0=>
    {"func"=>"FUNC_VNEG",
     "def"=>"tele(6),1120",
     "andsw"=>nil,
     "delay"=>50,
     "duration"=>0},

```

raphaelcoeffic, 2021-11-22:
It seems we have 2 choices:
- escape (something like `\!L2`)
- quote: `"!L2"`

Both seem to work with the YaML online linter I tried.

stronnag, 2021-11-22:
Note: This issue (lost !XX during round trip) applies to other tags as well, as might be expected.

Applying double quotes to the yml file fixes it (both ruby and python parsers).

raphaelcoeffic, 2021-11-22:
> Note: This issue (lost !XX during round trip) applies to other tags as well, as might be expected.
> 
> Applying double quotes to the yml file fixes it (both ruby and python parsers).

Ok, then let's double quote all of these.

raphaelcoeffic, 2021-11-27:
@stronnag while looking at fixing this, it appeared much easier to use `\!` instead of double quotes. Could you please re-try with this code if your Ruby parser likes it better?

stronnag, 2021-11-27:
Patched #1175 into current master. Neither the ruby nor the python YAML decoders report any issues reading any of 25 model files tests.

From (brief, visual) inspection:
* The python output had no obvious defects (but hard to be conclusive as it reorders the elements).
* In the ruby output i noticed:
  ```
  type: TYPE_MULTIMODULE
  subType: 30
  ```
where source is:
```
type: TYPE_MULTIMODULE
subType: 3,0
```
The python output was correct, so this is probably not a Edge issue.

PR #1175 (2021-11-28): YaML: escape exclamation marks to avoid issues with compliant parsers
Fixes #1131
closed 2021-11-28 by pfeerick