EdgeTX/edgetx #1133: Create some sort of virtual inputs settable fom LUA (like the new SHM variables for widget <-> mixer data exchange)

wimalopaan, 2021-11-22:
It would be real cool to create virtual inputs like sliders and gauges in widgets that produce input values.

I could imagie a way like the way it was done in #132 . Or just make the SHM accessible as input-sources.

wimalopaan, 2021-12-02:
@jfrickmann What is your opinion about this one?

jfrickmann, 2021-12-02:
Interesting idea, but also a lot more work than what we just did, because we would have to add something new to the system instead of just hooking into the existing.

jfrickmann, 2021-12-02:
Couldn't you do that with a Mixer script, now that you have the Shm tables?

wimalopaan, 2021-12-02:
Well, the easiest way would be to hijack a GV as I used to do that before SHM.
Perhaps your effort to increase the number of GVs (at least in colorLCD) is successful, then this request may not be neccessary anymore, But using GVs for this purpose still feels like hijacking ...

jfrickmann, 2021-12-02:
Yes, more GVs is the easy way to do it, assuming that it is as simple as increasing MAX_GVARS and ignoring all of the warnings. But I agree, it is not the most elegant solution, and it uses a lot of memory for what you get.

Maybe we could discuss if the functionality of Mixer scripts needs to be reserved for a special script type that is not enabled by default, or if we could make some general API functions that do the same thing, and can be used from widgets and other scripts. One problem with that approach, is that a certain input source will pop up and disappear at random as different scripts run, and that two different scripts may write to the same input. Maybe not so easy, it would certainly need careful analysis!

wimalopaan, 2021-12-02:
PLEASE: DON'T REMOVE MIXER-SCRIPTS FROM EDGETX. They are so usefull!!!

wimalopaan, 2021-12-02:
> Maybe we could discuss if the functionality of Mixer scripts needs to be reserved for a special script type that is not enabled by default, or if we could make some general API functions that do the [...]

I'm afraid that I did not excactly understand your suggestion.

jfrickmann, 2021-12-02:
When I look into the source code, I see evidence that Mixer scripts is a concept that has evolved over time. Sometimes it's called "mixer", other times "model" scripts. If you compile with -DLUA=NO, then there is no Lua, -DLUA=YES gives you everything including Mixer. But the default is -DLUA=ON, which is strange, and it gives you everything Lua _except_ Mixer... 

So clearly, there has been some debate about this concept, and I do not have any special insight here. But I think that some were concerned about users misusing this power to have scripts control planes directly, and I am guessing that's why they made it a non-standard build option.

There is a #define `LUA_INPUTS` that seems to be set for all ARM STM32 targets, and it is needed for mixer scripts. So the "plumbing" needed for Lua inputs is in place, even without mixer scripts. Maybe there has been some past debate about an alternative like what you bring up here - who knows?

But one way to cut it, could be to provide a fixed set of Lua inputs with fixed names (like you will see in Companion model edit when Mixer is enabled but no scripts are running). And have Lua API functions that can be used everywhere, also in widgets, to set these inputs. 

And have this enabled when Mixer is _not_ enabled, so it is not competing for the same space. This would be an alternative to Mixer scripts, and if it gathers general support, it would over time superseed Mixer scripts. But if you prefer to keep the custom build option with Mixer scripts, then it would not affect you at all.

I like this idea, because it does not add anything new to the underlying EdgeTX system, it just uses the existing infrastructure in a new way. And we get a nice and clean separation between old and new, because the old is already a custom build option.

Just a thought - please sleep on it before you respond 😄

wimalopaan, 2021-12-06:
> When I look into the source code, I see evidence that Mixer scripts is a concept that has evolved over time. Sometimes it's called "mixer", other times "model" scripts. If you compile with -DLUA=NO, [...]

Ok, I wasn't aware of this fact. Thank you for pointing that out (at least to me ;-) )

> So clearly, there has been some debate about this concept, and I do not have any special insight here. But I think that some were concerned about users misusing this power to have scripts control planes directly, and I am guessing that's why they made it a non-standard build option.

Mixer-scripts (I would prefer that name) are a very powerful tool. I strongly argue against removing them from the system at all. I see a strong parallel to the heli mixers! We could make both (heli-mixer and mixer-scripts) run-time de/selectable in the future. Heli-mixer are a very special tool for heli-pilots - no one else needs them. And mixer-scripts are very special and useful for e.g. ship- and crawler-modelists. So, they have to stay alive.

If I understand you correctly you are argueing towards selecting either mixer-scripts or LUA-inputs (from widgets oder telemetry). 
Well, I don't think that to be a good idea. It should be possible to use both.
 
I the two different use-cases here:

1) mixer-scripts should do a special calculation on the inputs only to get an outputs - like computing the outputs for e.g. Voith-Schneider-Propeller or Schottel-Thruster (conceptual the same as with heli-menu)

2) LUA-inputs should give the possibility to create some fancy (virtual) inputs e.g. some touch widgets to produce something that can be used as inputs like physical inputs now. 

And because global variables have their own use-cases already, there should be no need to hijack them. 

So, I propose the use of the newly created SHM-variables as general inputs (selectable from the UI). And - yes - this would eliminate the use of some simple mixer-scripts, that only transport a GV or SHM to some output.

jfrickmann, 2021-12-06:
So maybe we can get by with the existing features? Make a widget that writes values to `ShmVar`s, and a mixer script that reads the same `ShmVar`s and outputs them?

wimalopaan, 2021-12-06:
> So maybe we can get by with the existing features? Make a widget that writes values to `ShmVar`s, and a mixer script that reads the same `ShmVar`s and outputs them?

Yes, surely that is an option which I use.
But we can get rid of these one-liner mixer-scripts when we define the `ShmVar` s as inputs. You said above that the infrastructure for this to happen is allready there.

jfrickmann, 2021-12-06:
> You said above that the infrastructure for this to happen is allready there.

Yes, and it is used by the Mixer scripts, as I also pointed out.

Since you already have everything you need with `ShmVar`s and Mixer scripts, I do not think that we should add new features doing the same thing, albeit with a different API.

wimalopaan, 2021-12-06:
Ok, I'm not in the position to fix/add that by myself ;-) So I have to stick with the current solution. 

There is one major drawback: we can't setup mixer-scripts via lua itself. But we can setup mixer/mixer-lines. So, there is manual interaction needed to setup a full functional processing chain.

To achieve this, we need either

1) the possibility to insert mixer-scripts

or

2) the use of ShmVar's as mixer-inputs.

Well, if 1) would be possible, I'll be totally fine ;-)

jfrickmann, 2021-12-06:
1. is a completely different direction than this issue, and if you want that, then please open a new feature request issue, for possible implementation in a future EdgeTX version.
2. you can already do with a mixer script.

wimalopaan, 2021-12-06:
Already there: https://github.com/EdgeTX/edgetx/issues/931