EdgeTX/edgetx #1134: wrong color assignment in some widget elements, size of "Channels"

notyourbiz, 2021-11-22:
while playing around with themes i figured out that the bars in "channels" have a wrong color-assignment, like the ForeColor in Mike Shellim's "Showall" V 0.9.8 (latest)

My Theme:
[image]
There is no blue at all...

Widget "Channels"
[image]

Widget "Showall":
[image]

for comparsion same on OTX:
[image]

[image]

As U may also notice, in ETX Widget "Channels" is cut on the Bottom compared to OTX, even without Trims etc...

[image]

And the "Batt-Value" in the Top-Bar on ETX is black and not changeable, on OTX it's the same color like the rest in the Top-Bar, which looks like it should be... my 2ct...

Maybe someone else figured all that out long before me, if so pls close the case...

THX

notyourbiz, 2021-11-22:
sorry, forgot:

ETX 2.6.0
TX16S

eshifri, 2021-11-23:
The color of the "Batt" widget is configurable in the Widget settings.

pfeerick, 2021-11-23:
[image]

pfeerick, 2021-11-23:
ShowItAll

pfeerick, 2021-11-23:
A future option that I hope to add to all the built-in widgets is a 'use theme colours' (on by default) to all widgets, but need room for more than five options before that can happen ;)

notyourbiz, 2021-11-23:
i got all that, thx guys...

but to come back to colour assignments lets talk about the "showall"-widget...

> ShowItAll widget also has options to use default theme colours, and set fore and background colours.

as i mentioned, i have no blue at all in my theme (which is just experimental...) but the "showall" (set to dflt) picks  a blue ForeColor - where from? the BackColor is just picked right..

and what stand the values for color-settings of the widgets for? it's not RGB nor HSV. maybe i'm just to stupid, but to me its impossible go get a proper color.match (e.g. backgrounds Theme/Widget). do we need 3 types of color settings? RGB, HSV & ???

> Bottom line of outputs is being clipped by the flight mode display text (untick in screen options) to see the change - I'll see if I can move that by a row - as it's probably only a 1px overlap or something.

made my day again... ;-)

pfeerick, 2021-11-23:
Oh, very nice catch... it looks like lua font colour is probably not being set to PRIMARY1...

notyourbiz, 2021-11-23:
one more thing on ""outputs"-sliders-dflt.color:
of course its adjustable, i got that...

but it always starts with a blue #19232 (RGB565, i learned)

if you watch the "Channels Monitor", the Mixers-Sliders are assigned to FOCUS

wouldn't it make sense to assign the "Outputs"-Sliders to FOCUS too by default?

pfeerick, 2021-11-23:
Hm... it's actually supposed to be getting the defualt for the bars from `COLOR_THEME_SECONDARY1`... I'll have to see why it's not getting the active theme colours...

eshifri, 2021-11-23:
ShowAll.lua uses hard-code colorFlag=0 for default. If you change it to COLOR_THEME_PRIMAR1 you will achieve your goal.
local function refresh(wgt)

	-- Colour option
	-- Check for LS bit (Github #7059)
	if bit32.btest (wgt.options["Use dflt clrs"], 1) then
		**colorFlags** = 0
	else
		lcd.setColor (CUSTOM_COLOR, wgt.options.BackColor)
		lcd.drawFilledRectangle (
			wgt.zone.x,
			wgt.zone.y,
			wgt.zone.w,
			wgt.zone.h,
			CUSTOM_COLOR)
		lcd.setColor (CUSTOM_COLOR, wgt.options.ForeColor)
		colorFlags = CUSTOM_COLOR
	end

pfeerick, 2021-11-24:
> ShowAll.lua uses hard-code colorFlag=0 for default. If you change it to COLOR_THEME_PRIMAR1 you will achieve your goal. local function refresh(wgt)

I feel that in itself is a bug, as if the colourflag is 0, then the default colour should be used, and the default colour is always blue, instead of COLOR_THEME_PRIMARY1, regardless of theme colour. You can see the same thing happen if you do drawText without specifying a colour.

eshifri, 2021-11-24:
Is it relevant?
LcdFlags flagsRGB(LcdFlags flags)
{
  // RGB or indexed color?
  if (flags & RGB_FLAG)
    return flags;
  else
    return (flags & 0xFFFF) | COLOR(COLOR_VAL(flags)) | RGB_FLAG;
}
For flags==0  COLOR_VAL(flags)==0 and COLOR() is called with DEFAULT_COLOR_INDEX.

DEFAULT_COLOR_INDEX==0, while COLOR_THEME_PRIMARY1_INDEX==1

pfeerick, 2021-11-24:
Yes, I think this is what it comes down to in the end... `DEFAULT_COLOR_INDEX` being hard coded, and not controlled by themes. 
https://discord.com/channels/839849772864503828/842693696629899274/912884479557783562

notyourbiz, 2021-11-24:
tried the 78fc55c, but to me nothing changed...

- formatted SD
- complete untouched SD-Content (newly set)
- nightly 78fc55c in FIRMWARE
- untouched showall 0.9.8 in WIDGEDS
- flash from SD-Card
- manually setted new "Camo-Theme"
=> widgeds behave like before

sorry if i bother you...

eshifri, 2021-11-24:
It has not fixed, so you have bothered yourself. :-) You will receive an automatic  update when the fix will be delivered.

notyourbiz, 2021-11-24:
>  bothered yourself. :-)

never ever...

pfeerick, 2021-11-24:
Yes, that was an unrelated edit. There will be an update in this issue once it is resolved ;)

closed 2021-11-27 by pfeerick

DG9BFC, 2022-08-07:
found another bug .. color of top widget changes color depending what color a main widget is and if that is set to fullscreen ...
i have a few widget in main screen with different colors ... and i have two widget added in top row (tx and rx batt as gauge and with value) ... those two widgets in top row changes if i select one of the widgets in main and set it to fullscreen (and back)
the last "set to full" sets the color of the two gauges in the top row

PR #1148 (2021-11-27): Use COLOR_THEME_PRIMARY1 as a default lua script color
Fixes #1134 at least partially

Summary of changes: if lua script uses colorFlag==0 use PRIMARY1 color instead of default color.
The change effects lua scripts only.