EdgeTX/edgetx #1135: Appearance Trims & Sliders

notyourbiz, 2021-11-22:
On OTX they look ways better:

OTX:
[image]

ETX:
[image]

just my2ct, but the OTX version looks more proffessional to me...

MervDale, 2021-11-22:
Black with a bit of brown as opposed  to brown with a bit of black, I guess it's in the eye of the beholder.

notyourbiz, 2021-11-22:
what i meant is the fine-lined, round-endet trimbars (OTX) vs. the simple "stripes" in ETX.

and if i could have changed the brown stripes in ETX , i would have done...

pfeerick, 2021-11-23:
Rounded rectangle, I'm not sure, but I changing it to a empty rectangle can certainly be done, and I think would make the 'trim knob' a lot more visible as to where on the bar it is...

MervDale, 2021-11-23:
It seem s to me it's all cosmetic and not such a big deal to raise an issue on as it all depends on how you want to set you screen up, as I said in the eye of the beholde
[image]
r

pfeerick, 2021-11-23:
Ah, but you're not the centre of the universe, I'm afraid MervDale, I am :-P

Anyway, there will be no right answer there, as if you have a wallpaper like yours, the trim bars being filled actually make things more visible, whereas outlines would be near impossible-to see in some places. This is where options would be nice, since it really depends on the theme and background.

Eldenroot, 2021-11-23:
Also check the timers on the right, not aligned properly in edgetx. And one more - header top bar - color of battery voltage shoul be white

pfeerick, 2021-11-23:
> And one more - header top bar - color of battery voltage shoul be white

Colour is configurable ;)

> Also check the timers on the right, not aligned properly in edgetx

The latest nightly may be different - it has monospaced font for numbers

notyourbiz, 2021-11-23:
@MervDale as to be seen in your example the liders and trims have the same color than the top-bar... both are assigned to SECONDARY1

so, how to make S&T transparent without loosing the TopBar?

obviously i'm to stupid...

Mike-HRC, 2023-05-18:
Did anything happen with this please?

I find the transparent bar on OpenTx much easier to see than the solid bar of EdgeTx

MervDale, 2023-05-19:
It is still the same option trims, sliders and top bar are the same colour, maybe you could see if a differnt theme done by all the other people suits your needs better, with my full colour background I find solid bars more visible but then I only use the ETX default theme. cheers

Mike-HRC, 2023-05-19:
Thanks Merv, I did have a look at some others and even tried tweaking the settings on EdgeTx Theme Creator, but I just find them a little "clunky" with the solid bars