EdgeTX/edgetx #1138: in the last nightly DSC port doesn't work

tonims1, 2021-11-23:
In the lastest nightlies It have no ppm output on the top DSC port, 
Simulator does not work for me

**General information**
 - EdgeTX version (latest nightlies )
 - Hardware (RMTX16S)

raphaelcoeffic, 2021-11-24:
@tonims1 the TX16S has quite some issues with the physical connection. You might want to have a look at that. @rotorman has described some of the issues on RCG.

pfeerick, 2021-11-24:
Actually, looks like something may have happened... DSC simulator cable is working with 2.5.0, but not a nightly from probably two days ago. er... the trainerData block don't look right... 

```yml
trainerData: 
   mode: SLAVE
   channelsStart: 0
   channelsCount: 0
   frameLength: 0
   delay: 0
   pulsePol: 0
```
[image]

raphaelcoeffic, 2021-11-24:
Ouch!

pfeerick, 2021-11-24:
Well that makes sense, given when I moved everything one value off they changed +1 or -1 ... but DSC not working does appear to be firmware related ;)

rotorman, 2021-11-26:
Problem confirmed with RM TX16S (analog sticks) and today's nightly. With Trainer set to Slave/Jack this is what I see on logic analyzer (does NOT resemble PPM):

[image]

(bottom line is the same signal, just sampled with analog circuitry).

raphaelcoeffic, 2021-11-27:
Fixed by 453e478

closed 2021-11-27 by raphaelcoeffic