EdgeTX/edgetx #1139: TX16S TQly and TRSS telemetry missing

rburrow87, 2021-11-24:
I noticed that the TQly and TRSS telemetry sensors were suddenly dead in my models using FrSkyX2 D16 the last couple of days. I confirmed that something changed after Nov 20th because I flashed to a bin I compiled on the 20th and they are working again. Flashed back to a current nightly and they stop working again.

eshifri, 2021-11-24:
Can you please try to run sensors rediscovery?

rburrow87, 2021-11-24:
I should have mentioned, I did try deleting sensors and re-doing sensor discovery and they would not come back without flashing the firmware.

pfeerick, 2021-11-24:
Thank you both... I meant to ask that yesterday. Ok, I'll try this locally - I got half-way there before getting distracted.... but something has probably changed in yaml to trip this up.

eshifri, 2021-11-24:
There were also some changes in modules and telemetry. I have briefly looked into it and had not found anything suspicious, but...

rburrow87, 2021-11-26:
This may be another clue. I was messing around with it more and I noticed that on the "good" build, if I'm using the internal multi module and I set the external module to something like CRSF and leave the internal on, TQly and TRSS break again.

pfeerick, 2021-11-26:
Right, so binding an X8R via MPM on TX16S with todays nightly only yielded this:
[image]

Rolling back to 2.5.0, and configuring the MPM/rediscovering sensors yielded:

[image]

koaxheli, 2021-12-19:
Has not changed anything in the telemetry to date.
TRSS and TQly still not recognized.
[image]

pfeerick, 2021-12-21:
I wonder if this is related to #1039 as that was from just around then.

rburrow87, 2021-12-23:
I just noticed today, if I use an external multi module (Jumper JP4IN1 v1.3.3.0), they come back and work with the external module.

raphaelcoeffic, 2022-01-18:
@rburrow87 Can you please confirm / infirm whether 1) this still happens, and 2) if this is only with the internal ISRM and/or MPM?

rburrow87, 2022-01-18:
Checked with 2.7 nightly 7da7261e on my TX16S and it's still happening with the internal Multi. I've only seen these sensors with the MPM, not with the ISRM.

If I set the external RF to Multi, they start working again even if the RX is only connected to the internal module.

pfeerick, 2022-01-19:
This gels with my earlier comment re external RF setting also messing with those sensors on 2.5, so it seems like we have a long-running issue that has been recently exacerbated.

richardclli, 2022-01-21:
I have a friend used the RM receiver R161, and used the S.Port to send back telemetry, the included figures are from OpenTX 2.3.9 and compared to EdgeTX 2.6, EdgeTX has a lot of sensors missing.  And he tried EdgeTX 2.5.0 as well.  Similar result, a lot of sensors missing.

I am thinking if the Frsky Hub telemetry is broken from day 1 in the OpenTX 2.4 branch?  Maybe we need to compare the current OpenTX sources to see what happens.

[image]
[image]

raphaelcoeffic, 2022-01-21:
@richardclli how is the R161 connected? We don’t poll the sensors, it’s the module’s job.

pfeerick, 2022-01-21:
Hm... I wonder if this has connection in some way... https://github.com/EdgeTX/edgetx/issues/1260

commit a7d3927c (2022-01-22): multi: move the TQly & TRSS hack where it belongs

Resolves #1139
PR #1480 (2022-01-23): multi: move the TQly & TRSS hack where it belongs
Resolves #1139
closed 2022-01-23 by pfeerick