EdgeTX/edgetx #1141: LUA table.* library for NV14

JimB40, 2021-11-24:
Seem like LUA table.* library was added long ago to HORUS tagrets (thx @jfrickmann for finding) even luadoc states is not.
https://github.com/opentx/opentx/pull/7599
Still doesnt work on NV14

Looked at code and seems to me it's here.
https://github.com/EdgeTX/edgetx/blob/78fc55c06d175d0b6716215d2ff50819c0064aca/radio/src/thirdparty/Lua/src/linit.c#L56

btw @raphaelcoeffic adding this to color radios only is about memory usage?

pfeerick, 2021-11-24:
It would be about runtime memory *AND* firmware size

JimB40, 2021-11-24:
Still with NV14 should be no issue I think.
And that's funny because it will make LUA more compact on targets with more memory:)

raphaelcoeffic, 2021-11-24:
I think the `#if defined` can just changed to `COLORLCD` instead of just `HORUS`.

PR #1147 (2021-11-25): Enable lua `table`  for colorlcd, not just horus style targets
Resolves #1141 

Summary of changes:
- Enable lua table library for all colorlcd targets, not just Horus-style targets
- Firmware size change: 1397888 => 1402632

Tested with following on both TX16S and NV14 hardware. 
```lua
local toolName = "TNS|Table Sort Test|TNE"

local function run(event)
    local t = {5,4,3,2,1}
    local w, h = 0, 0
    local x, y = 0, 0

    -- Lazy B&W test
    if LCD_H > 64 then
        w, h = lcd.sizeText("M")
    else
        w, h = 8, 8
    end

    lcd.clear()
    
    x = 5
    lcd.drawText(x, h*1, [... cut]
closed 2021-11-25 by pfeerick