EdgeTX/edgetx #1144: EdgeTX Companion Windows 2.5.0 error 0xc000007b

julianschweizer, 2021-11-24:
I have trouble running EdgeTX Companion Windows 2.5.0. After starting nothing happens. I found this in the event log:

`- <Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
- <System>
  <Provider Name="Application Popup" Guid="{47bfa2b7-bd54-4fac-b70b-29021084ca8f}" /> 
  <EventID>26</EventID> 
  <Version>0</Version> 
  <Level>4</Level> 
  <Task>0</Task> 
  <Opcode>0</Opcode> 
  <Keywords>0x8000000000000000</Keywords> 
  <TimeCreated SystemTime="2021-11-24T16:14:12.2448219Z" /> 
  <EventRecordID>8454</EventRecordID> 
  <Correlation /> 
  <Execution ProcessID="900" ThreadID="1664" /> 
  <Channel>System</Channel> 
  <Computer>DESKTOP-KCQ16CH</Computer> 
  <Security UserID="S-1-5-18" /> 
  </System>
- <EventData>
  <Data Name="Caption">companion-windows-2.5.0.exe - Anwendungsfehler</Data> 
  <Data Name="Message">Die Anwendung konnte nicht korrekt gestartet werden (0xc000007b). Klicken Sie auf "OK", um die Anwendung zu schließen.</Data> 
  </EventData>
  </Event>
`

Prozessor	Intel(R) Core(TM) i7-8700K CPU @ 3.70GHz   3.70 GHz
Installierter RAM	32,0 GB
Systemtyp	64-Bit-Betriebssystem, x64-basierter Prozessor
Edition	Windows 11 Pro
Version	21H2
Installiert am	‎27.‎10.‎2021
Betriebssystembuild	22000.348
Leistung	Windows Feature Experience Pack 1000.22000.348.0

Any ideas?

Nerenye, 2021-11-25:
0xc000007b is usually a .net Framework error and has nothing to do with windows11. But i´m not quite sure why that would pop up on companion

Try to reinstall the latest vc++ redist

julianschweizer, 2021-11-25:
Strange, vc++ redist all deinstalled and re-installed latest release. No luck, same „0xc000007b“ error.

Nerenye, 2021-11-25:
Are you on the latest Build and also on the release channel?

julianschweizer, 2021-11-25:
Yes it is, 21H2 Build 22000.348 and all updates installed. I also tried to reinstall .NET Framework. Does not help.

Nerenye, 2021-11-25:
so youre not on the release channel. Official Build is 
[image]

julianschweizer, 2021-11-25:
You are right I accidentally installed the preview. Manual removal resulted in a complete crash … .

 Thanks for the help. :)

rapuyrav, 2021-12-07:
I have the same problem on Windows 10.
OpenTx Companion is working fine, but I'm not able to install EdgeTx Companion with the exact same behavior as above.

I've try everything I could:
- reinstall vcredis countless times (every version from 2008 to latest),
- using visual studio installer to install C++ dev tools,
- reinstall visual studio,
- compile myself.
None of this worked, as I was not able to install MSYS2 for the same reason I think.

Fun fact: I can install it fine on Windows Sandbox !
I can't figure it out what's causing this on my main system.

Specifications:
Edition	Windows 10 Pro
Version	21H2
Installed on	‎06/‎01/‎2021
OS build	19044.1348
Experience	Windows Feature Experience Pack 120.2212.3920.0

julianschweizer, 2021-12-07:
I was on a W11 preview, trying to roll back manually broke my system. After fresh install without preview release it worked fine.

elecpower, 2022-11-16:
@julianschweizer @rapuyrav @Nerenye is this still an issue or can we close?

Nerenye, 2022-11-16:
I can't say I did not have that problem. Just tried to help

julianschweizer, 2022-11-16:
Close. New Windows Installation helped.

elecpower, 2022-11-16:
Thanks for the update. Closing

closed 2022-11-16 by elecpower