EdgeTX/edgetx #1152: Receiver name does not appear in the bind menu

Skippydi, 2021-11-25:
When binding, the receiver name does not appear in the registration menu or in the bind menu
Hardware Horus 10s with ISRM Modul
OpenTX Version 2.6.0 (d775e75e) main version

pfeerick, 2021-11-25:
I think you meant EdgeTX there ;)

Skippydi, 2021-11-26:
Sure

raphaelcoeffic, 2021-11-26:
Just to be sure: you power the registered receiver after entering the bind menu, right?

Skippydi, 2021-11-26:
Yes, of course.

ChrisOhara57, 2021-12-28:
interesting fact.
By any way, have you swapped the receiver from one model to a new programmed model??

I do ask because I have the same effect on my X9liteS but I'm using   openTx 2.3.11 (Access)
I have 3 receiver which I do use in two models each.
The names of the receivers (rx6r, gr6 and gr8)  only display in the original model and not in the new created model ([Bnd] instead)
But the receivers work fine in all models.
But when I copy a model the receiver name shows up in both models new and old.

raphaelcoeffic, 2022-01-02:
> In the second model I had to register once more but with the same registration ID as in the first model. (changed manually) Then turned the receiver off and did a new bind this time the receiver name popped up and could be selected.

Does that mean the error is actually in the RX firmware? (not transmitting stuff when not registered with that model ID?)

ChrisOhara57, 2022-01-02:
> Does that mean the error is actually in the RX firmware? (not transmitting stuff when not registered with that model ID?)

No; As said transmitting is ok, I flew with this three receivers about have a year.
       But in the menu the receiver name does not appear  instead it shows [BND] where [RX6R] should be.
       Also you should be able to select the receiver name and get an option menu, that is not possible in the second model.
       But transmitting works.
       The origin of the problem seems to be that I either bound or registered the receiver in a second model on the 
       same transmitter.
       Probably I should have taken care to use the same ID manually.
       (it is have a year ago now that I did the bind) 
       I do not know how the procedure works but is it possible that either the transmitter or the receiver does choose a 
       different ID in case of a second model. (In all other models the procedure automatically used the same ID)