EdgeTX/edgetx #1153: User interface: pages don´t snap in using touch

3DRacer, 2021-11-25:
Describe the bug
If you have set several pages (mainview 1, mainview 2, ....) in the user interface, the individual pages are no longer self centered when using the horizontal swipe.

To Reproduce
set more than one page in user interface. Swipe through the pages using touchscreen. The view stops also between pages.

Expected behavior
every page should snap in whole view when the screen is not touched.

General information

    EdgeTX version 2.6.0 d775e75
    Hardware RM TX16S

pfeerick, 2021-11-25:
@gagarinlg This is part of what I was meaning before with slide being more snappier than before, and inertia not being as smooth.

gagarinlg, 2021-11-26:
> @gagarinlg This is part of what I was meaning before with slide being more snappier than before, and inertia not being as smooth.

I think I found it. I only have to check, if the fix affects the touch off when backlight is dimmed function.

gagarinlg, 2021-11-26:
this is fixed by [libopenui #38](https://github.com/EdgeTX/libopenui/pull/38)

closed 2021-11-26 by pfeerick