EdgeTX/edgetx #1159: Make CRSF option in cmake work again (-DCROSSFIRE=OFF) (on T12)

wimalopaan, 2021-11-26:
With lastest main it is not possible to deconfigure CRSF (on T12, maybe others).

```
cmake -DUSB_SERIAL=OFF -DHELI=OFF -DTRANSLATIONS=DE -DPCB=X7 -DPCBREV=T12 -DINTERNAL_MODULE_MULTI=OFF -DLUA=YES -DGVARS=YES -DGHOST=OFF -DAFHDS3=OFF -DCROSSFIRE=OFF -DOVERRIDE_CHANNEL_FUNCTION=OFF ..
```

gives

```
/home/lmeier/Projekte/edgetx/edgetx/radio/src/pulses/pulses.cpp: In Funktion »void intmoduleSendNextFrame()«:
/home/lmeier/Projekte/edgetx/edgetx/radio/src/pulses/pulses.cpp:746:47: Fehler: »union InternalModulePulsesData« hat kein Element namens »crossfire«
746 |       intmoduleSendBuffer(intmodulePulsesData.crossfire.pulses,
|                                               ^~~~~~~~~
/home/lmeier/Projekte/edgetx/edgetx/radio/src/pulses/pulses.cpp:747:47: Fehler: »union InternalModulePulsesData« hat kein Element namens »crossfire«
747 |                           intmodulePulsesData.crossfire.length);

```

raphaelcoeffic, 2021-11-26:
Ok, the issue is that by using `-DCROSSFIRE=OFF`, you assumed it would disable the capability for CRSF as internal module. But this is not how it works. However, it was not really configurable, which is why I made #1161.

raphaelcoeffic, 2021-11-26:
@wimalopaan but what it is actually you are trying to achieve, when disable MULTI, CRSF, GHOST and AFHDS3? What is actually left?

wimalopaan, 2021-11-26:
Well, I only need MPM, and that works (as external module).

raphaelcoeffic, 2021-11-26:
> 

So no internal modules at all??? Do you have one these very old T12 without internal module?

wimalopaan, 2021-11-26:
Excactly: Jumper T12

wimalopaan, 2021-11-26:
Just tested only with `-DINTERNAL_MODULE_CRSF=OFF` and that is ok now.

PR #1161 (2021-11-28): Make list of supported internal module configurable per cmake option
Resolves #1159

This allows for configuring the available internal modules via cmake as follows:
```
cmake -DPCB=X7 -DPCBREV=T12 -DINTERNAL_MODULES="PXX1;PXX2" -DGHOST=OFF -DAFHDS3=OFF -DCROSSFIRE=OFF ...
```

Please note that `-DCROSSFIRE` will disable the CRSF on the external module interface. Without removing CRSF from the internal modules, it would automatically activate support for CRSF, and thus override `-DCROSSFIRE`.
commit b8335b99 (2021-11-28): Make list of supported internal module configurable per cmake option

Resolves #1159
closed 2021-11-26 by wimalopaan