EdgeTX/edgetx #1170: Special Function Action as Lua Script should be either repeat or once

happy12, 2021-11-27:
Considering the following simple lua script:
```
local function run()
  playFile("/SOUNDS/en/system/expo.wav")
  playNumber(20, 13)
  return 1
end
return { run = run }
```
When used in a Special Function as an Action "Lua Script" (or "Play script") it is constantly repeated. The user should be able to set it for play once, including play once, not during startup. Or if no option, the default should be "once".

**To Reproduce**
Set any lua script as a Special Function, it will repeat itself.

**Expected behavior**
Not repeat itself by default. The user should be able to write a script with a while loop if repetition is desired.

**General information**
 - EdgeTX version 2.6.0 (22112de3)
 - Hardware TX16S

pfeerick, 2021-11-27:
Except this is the expect behaviour of a Lua Function script (triggered by a SF). Use a combination of the init, run and background functions if you want to change it to be 'single shot' per trigger. 
https://luadoc.edgetx.org/part_i_-_script_type_overview/function_scripts

closed 2021-11-27 by pfeerick

lshems, 2021-11-27:
I don't know if this is preserved in edgetx, but in opentx the BACKGROUND function is called periodically when the switch for executing the function becomes false, and RUN when true.

This is allows to control the behaviour on the special function completely. 

Declare a local variable 'count' outside the run and background function, so both functions 'know' it ( as an upvalue.).

Set the run to execute the play if the counter is zero, and then set counter to 1.

Set the background to set the counter to 0.

That's how this is intended to work.