EdgeTX/edgetx #1184: SFs Yaml Questions/Feedback

pfeerick, 2021-11-28:
Right, so I loaded up all the special functions on the latest nightly, and they all seemed to have behaved. Just one or two minor "should it look like that" questions - nothing... functionally wrong. For your viewing pleasure ;)

- [ ] SF: "Play Sound"  Um... `FIRST_WITHOUT_ENABLE` was indeed my first choice in what it should be called!?
```yml
   10:
      swtch: SD1
      func: FIRST_WITHOUT_ENABLE
      def: Bp1,3
```

- [ ] SF: Adjust => Global var... since parameter 1 is GV1, does parameter 3 (which can only be a GV) need to be wrapped with `gv()`?
```yml
   7:
      swtch: SC1
      func: ADJUST_GVAR
      def: 0,GVar,gv(3),1
```

- [ ] SF: Screenshot, Vario, Racing Mode and Disable touch don't have any options - do they need a def block? 
```yml
   0:
      swtch: SH2
      func: SCREENSHOT
      def: 
   14:
      swtch: SE2
      func: VARIO
      def: 
   17:
      swtch: SG2
      func: RACING_MODE
      def: 
   18:
      swtch: SF2
      func: DISABLE_TOUCH
      def: 
```

raphaelcoeffic, 2021-11-28:
> `FIRST_WITHOUT_ENABLE`
This is generated from the `enum`, I'd prefer not to change, or we need to change the enum as well (so in the code everywhere).

> since parameter 1 is GV1, does parameter 3 (which can only be a GV) need to be wrapped with gv()?

Yeah, makes sense, we can change that.

> SF: Screenshot, Vario, Racing Mode and Disable touch don't have any options - do they need a def block?

It could probably be changed, but that would require serious extension of some generator structures. This can be IMHO done later, as it does not really change anything functionally. Everything empty can be left out when generating in an another software.

pfeerick, 2021-11-28:
Oakley dokley, sounds good. 

Here's the crazy thing with 1) - my preferred choice is actually right there, and is the name used internally... except for... you guessed it! 🤦 

https://github.com/EdgeTX/edgetx/blob/9a5663e6114e3b954e7207cdf074d9cac517aada/radio/src/dataconstants.h#L815-L819

raphaelcoeffic, 2021-11-28:
> 

Oh! That is easy then! Just add `SKIP` after the definition for `FUNC_FIRST_WITHOUT_ENABLE`.

pfeerick, 2022-01-23:
Was basically closed in #1248

closed 2022-01-23 by pfeerick