EdgeTX/edgetx #1185: X10/S: Antenna configuration options missing

pfeerick, 2021-11-28:
**Describe the bug**
On OpenTX, there was an option to configure the Antenna options in RADIO/SYS -> Hardware. Values ranged between Internal, Ask, Per Model, External. Defaulted to "Per Model". This is missing. 

If Per Model selected, there was an option under 'XJT' in the Model -> Internal RF section for Antenna, which allowed choice of Internal, Ask, External. It appears that this is missing also. 

**Screenshots**
[image]
[image]

**General information**
 - EdgeTX version 2.6.0 nighties
 - Hardware: Frsky X10/S

brainbubblersbest, 2022-02-19:
Same on x12s. The last thing missing in access integration on edgeTX. 🙂

maxxam4, 2022-03-20:
will there be a solution in 2.7 Black Pearl? 

EdgeTX version 2.7.0 nighties
Hardware: Frsky X12S

pfeerick, 2022-03-21:
Probably not. I was working on code for this though, so I'll see if I can find it again and see what I got stuck on...

maxxam4, 2022-03-21:
Thanks for the information. It would be nice to be able to switch antennas for the open air season.

GWguzzi, 2022-04-19:
But one question, what is the current setting, is the external antenna output switched off?
May be it is a problem if no external antenna is connected at the Horus 12S while the output is switched on!

pfeerick, 2022-04-19:
The current setting would depend on what you set it to prior to using ETX... since the setting would be migrated across. 

It seems from what Companion is default to that it defaults to internal - but you can always read your transmitters current settings in Companion and check - the setting is still there, and Companion should read it. 

[image]

closed 2022-10-11 by pfeerick