EdgeTX/edgetx #1187: R9M/FLEX colorlcd has missing options

pfeerick, 2021-11-28:
In enabling FLEX support as part of #1043, it was noticed that there is no "S.Port Link Enabled" shown at any time or RF Power option when FLEX selected. 

[image]
[image]

pfeerick, 2021-11-28:
> Tested and working on NV14 and T16. I can now bind to models using R9 FLEX firmware. Still missing the option for the S.Port link and the power setting but good start. 

https://github.com/EdgeTX/edgetx/pull/1043#issuecomment-975771809

DamienPlo, 2022-01-17:
I confirm after testing 2.6 RC3 on my TX16S, no power setting available in flex mode after selecting 915 or 868 Mhz. So I don't know what output power is by default.

federikk, 2022-02-01:
I'm appendig this one here, don't know if it's actually a separate issue: 
**It must be selected FCC mode on the TX in order to have any output from the RX toward the FC**. 
Even if registration and binding has been obtained using EU or 868.
RcLink and telemetry can be estabilished instead using any band(FCC/EU/915/868). Outputs on BF/iNav rclink page are stuck but not in "no pulse" status/triggering failsafe, simply not moving.  Setting the radio to FCC solves the problem but i can't measure at which frequency (868 or 915) and maxpower (overlapping to this 1187).
Found with Tx16s, R9M2019, 2.6.0 custom externalaccessmod, F-port protocol, R9mm-ota, R9 Slim+ ota, betaflight 4.3, iNav 4.0

Arvycka, 2022-04-20:
R9M Flex power option is still missing in 2.7, but is available in the edgetx companion as well as inside the model yaml file. But is the value set there even used by the radio/module?

gagarinlg, 2023-10-14:
Ths S.Port link option is still missing in main

raphaelcoeffic, 2023-10-14:
> Ths S.Port link option is still missing in main

This is just a status display, not an option. Here is the code in OpenTX:
[image]

We have now some better code as shown in #4198 (`modulePortHasRx(module)`).

commit ff27271f (2023-10-14): feat: add S.PORT status for external R9M module

Fixes #1187
PR #4198 (2023-10-15): fix: MPM module status
We should only consider whether or not the MPM has a working telemetry link.

Fixes an issue introduced in #3822 

Fixes #4197, resolves #1187
closed 2023-10-15 by pfeerick