EdgeTX/edgetx #1188: X10S ACCST: OneBit support

BloatedGoatFPV, 2021-11-28:
On ACCST X10S Horus radio, I get repeating telemetry lost telemetry recovered.  I switched to Edgetx because of the OneBit feature but there is no Sample Mode in the Hardware menu for the Horus.

Because to my knowledge this problem affects all ACCST radios I was hoping the Horus was included in the OneBit solution.

EdgeTX version 2.5
Frsky Horus X10S ACCST, with Happymodel ELRS module.

pfeerick, 2021-11-28:
Since OneBit was your baby @hydra, what are your thoughts on this?

hydra, 2021-11-29:
I don't recall anything specific, but if the inverter is also bad on the Horus then it should be applied there too.  Perhaps @raphaelcoeffic can poke a config/build file somewhere to enable it for the Horus?

pfeerick, 2021-12-02:
Note to self about PR: https://github.com/EdgeTX/edgetx/pull/59. I *think* all that will be needed is for the options to be added to the colour GUI also https://github.com/EdgeTX/edgetx/blob/3ab896ba2ed0e37a26f4f3359fd036ec19b29abc/radio/src/gui/colorlcd/radio_hardware.cpp#L358  at this line so it's ins sync with the B&W GUI. Probably also data structure size and corresponding build tests updated also.

j-brubaker, 2021-12-19:
My Horus X10S ACCST on EdgeTX 2.5.0 also has trouble unless I lower the baud rate to 115K.  Tested on my Frsky R9 ELRS 2.0 module.

moyouyilansefengye, 2021-12-23:
I have an X10s radio，and want this amazing onebit mode sooo much. Havn't done any crsf  hardward mod yet. But please add onebit to X10. 
It would be so kind and nice for you :)

abimaputra78, 2021-12-27:
I had telemetry warning on radio master Tx16 s too. I had no one bit option. 🙏

pfeerick, 2021-12-27:
One-bit isn't an option of colourlcd radios generally, and specifically the RM TX16S does not need it. OneBit is a specific mode for older radios. If you are using ELRS, the only think you need to make sure of is that the max serial buad rate setting (which is in the radio/sys hardware tab near the bottom is set to 400k.

wucke13, 2022-01-09:
> OneBit isn't an option of colourlcd radios generally

That means that the X10S can not have OneBit mode, thus anyone experiencing difficulty with more than 115k baud on that radio has only one way of allowing more, the inverter mod?

raphaelcoeffic, 2022-01-09:
> > OneBit isn't an option of colourlcd radios generally
> 
> That means that the X10S can not have OneBit mode, thus anyone experiencing difficulty with more than 115k baud on that radio has only one way of allowing more, the inverter mod?

That is what it means for now.

wucke13, 2022-01-09:
What means "for now"? Is this a hardware limitation or a software problem? If it's the former, why don't we close this issue as its impossible to do?

raphaelcoeffic, 2022-01-09:
> What means "for now"? Is this a hardware limitation or a software problem? If it's the former, why don't we close this issue as its impossible to do?

We'd need some tests to see if it is really useful, so we don't give people false hopes. Apart from that, this is basically just a simple UI field. If it seems useful, it could be enabled rather easily.

wucke13, 2022-01-09:
I'm willing to test. I do have an X10S, a happymodel TX24 and I can only run it on 115k baud, so I also do have the setup to verify if it is any good.

j-brubaker, 2022-01-09:
> > > OneBit isn't an option of colourlcd radios generally
> > 
> > 
> > That means that the X10S can not have OneBit mode, thus anyone experiencing difficulty with more than 115k baud on that radio has only one way of allowing more, the inverter mod?
> 
> That is what it means for now.

I think what @pfeerick was saying there was specifically addressing the question described by @abimaputra78 with his TX16S and not making a statement about all color LCD radios.  The TX16S does not _need_ OneBit, so he was saying this GH issue isn't related to that specific problem.

The Horus X10S ACCST definitely needs OneBit, even though it is a color LCD radio.  Some, like the TX16S mentioned above, don't need it and it would likely not fix the telemetry warning issue described by @abimaputra78.

wucke13, 2022-01-09:
@j-brubaker Ah, that makes sense! I can't wait to try One-Bit mode, I'm definitely looking towards to not heaving to hardware mod a otherwise fine radio when the spare main-board is > 150$.

hydra, 2022-01-14:
It feels like there should perhaps be some configuration open to enable the feature only if the hardware has a slow inverter, and then support should be added to the horus which also has a slow inverter.  That way users of such radios will learn about the issue from their UI and other users won't have to even care about the setting.

I'm in favor of perhaps renaming the feature to 'Slow Inverter Workaround' or something, to users 'OneBit' is a bit (heh) cryptic.

wucke13, 2022-01-14:
> I'm in favor of perhaps renaming the feature to 'Slow Inverter Workaround' or something, to users 'OneBit' is a bit (heh) cryptic.

I gently oppose, the term is already coined. But maybe we could add on it, for example `Module Serial OneBit Mode`, to make clear that this relates to some serial device.

Specifically with the very high baud-rates that are now possible, wouldn't it be a sane choice to just offer the option to all users on all radios? At least retrospective I found it quite annoying to find the problem, than the claim that EdgeTX has a potential software fix only to learn that it was overlooked that my radio model could also be affected.

j-brubaker, 2022-01-21:
Is this on the roadmap for 2.6?  Not pushing, just wondering.

ChasedMonkey, 2022-05-21:
Hi, is there any update on this? I'd really like to use 400k baud on my x10s.

ChasedMonkey, 2022-05-23:
Yesterday with the help of Risto I ran a test build of 2.8 master with the "sampling mode" option enabled. 

Running one bit mode did let me run the elrsv2.lua at 400k baud where it wouldn't run at all with onebit turned off. But when running the lua at 400k baud it takes about 4x as long to open compared to 112k. 400k baud does not fix the constant telemetry lost/telemetry recovered messages, even if you run 250hz update. The lua functions the same at 112k baud regardless of One bit mode being enabled.

I believe the hardware mod is required for the x10s from my testing, I assume this is the same for the x10 due to the only change being the gimbals.

I have a hardware mod arriving today, once installed and can see if onebit mode has any benefits with the hardware mode, but I'm almost positive there will be no reason to use it.

BloatedGoatFPV, 2022-05-23:
Thanks for following up on this.

Doug aka BloatedGoatFPV

rotorman, 2022-05-24:
As a reference, here the branch of my fork with OneBit option added also to color radios: https://github.com/rotorman/edgetx/tree/onebit_color
Here the changes relative to vanilla EdgeTX main branch: https://github.com/rotorman/edgetx/commit/b40f38730e2ca75b5e3f9c68fa452e133a2d47c4

j-brubaker, 2022-08-26:
I'm still experiencing major "telemetry lost" warnings (and failsafes, though that confuses me) when running a Ghost module on my Horus X10S on 2.7.0.

I also tested the Ghost module itself in a QX7 with the inverter hardware mod (running OpenTX 2.3.13) and got zero telemetry errors.  I moved the Ghost module to a second Horus X10 running OpenTX 2.3.something, and with the baud rate manually lowered to 115k in the Hardware menu, I got zero telemetry errors.  I then tested a second Ghost module in all the conditions and got the same result.

I think the problem may lie in the fact that the X10S requires the inverter mod just like the QX7, but, according to radio/src/targets/horus/hal.h, the Horus (being RADIO_FAMILY_T16) is being assigned SPORT_MAX_BAUDRATE = 400000.  Then the Ghost telemetry driver is running at symmetrical speed (https://github.com/EdgeTX/edgetx/blob/dc99dd40d274df045b427f2fb7ab0968deaf9f94/radio/src/telemetry/ghost.cpp#L96) and causing problems with the X10's slow inverter.

Possible solutions would be to allow setting the baud rate manually when using the Ghost protocol or enabling OneBit mode as an option for the X10S and (and possibly the X12 as referenced in [#1949](https://github.com/EdgeTX/edgetx/issues/1949)).

Theoretically, these should be unrelated, but I've also observed failsafes in numerous conditions that make me wonder if there is a larger problem with the Ghost driver (maybe the addition of 12-bit resolution?) as it pertains to "slow" radios.  I don't have any hard data in that vein, so I won't speculate too much.  I don't think I had these issues on Edge 2.5, but as the Horus is my main radio, I'm a little reticent to flash backward to test on my daily driver.

I'm not knowledgeable enough to see what all is happening in the code.  But this is just what I've observed and dug out.  I'd be willing to test @rotorman's fork where OneBit is added to color radios, but compiling EdgeTX is going to be a little challenging for me.

gagarinlg, 2022-08-26:
Can you please make a backup of your SD Card and test the latest nightlies? There has been some work done in areas that might affect your problem.
Please be aware that the nightlies are not perfect, so you should stick to some ground testing. 
Regardless of the test with nightlies, you should update to 2.7.1, as you are missing some bugfixes when you are using 2.7.0.

j-brubaker, 2022-08-26:
Yes, I'll upgrade to 2.7.1 and test on my daily Horus.

I've been trying to sell my backup Horus, but I guess while I have it I can flash it with the nightly and test that as well.  I'll report back.

pfeerick, 2022-09-08:
Just to add more fuel to the fire :P

JC from OpenTX said about two years ago in relation to Ghost on the X7 
> You need to set baud rate to 115k, the channels will be sent at 400k, and telemetry at 115k. There is no performance to be gained by modding the radio

> the inverter mod "is useless and irrelevant for ghost module since we have introduced asymmetrical speed for ghost (the speed limit is only for data going in, not out)"

> if your Q7 allows it, either because it was factory fitted with fast inverter, or because you applied TBS mod already, then you can set it to 400k. In this case, both channels and telemetry will be transmitted at 400k, but it results only in more iddle time on the radio, it does not increase performance, latency, or throughput in any way for ghost (which is not true for GHST however) 

What part of the world are you in btw ;)

j-brubaker, 2022-09-08:
I'm in the U.S.

I see the code for the asymmetrical speed, but I believe that because the X10S is identified as "RADIO_FAMILY_T16", that code never runs and the Ghost driver runs instead at 400/400 (instead of 400/115).

Hopefully, the option of OneBit fixes the issues (as per [#1994](https://github.com/EdgeTX/edgetx/pull/1994)), but I suspect this particular case could also be fixed by setting "SPORT_MAX_BAUDRATE" to less than 400K, which would trigger the asymmetrical code.  

The best option might still be OneBit on color radios as the X12 might need it too.

Will OneBit appear for all External Module protocols?  I also have a Crossfire module but I've not tested it in the X10S.

j-brubaker, 2022-09-13:
Is this in the nightlies now (since it shows a 'merged' tag)?  I can repeat my previous test, if so.

pfeerick, 2022-09-13:
Yes, it has been in the nightlies since after 22 July ;)

bigjoncoop, 2022-09-20:
1st I just want to say thank you to all the devs and contributors for working on this.

So just to clarify, running edgeTX / OneBit mode on the Horus X10 does not help with 400k baud on external modules. For me, specifically a crossfire module.  

So the inverter mod is still the only way to fix this?

Thanks

pfeerick, 2022-09-27:
If it does not improve things in your specific case, then yes inverter mod is your only option AFAIK. For some radios/module combinations, it will work, for others, it doesn't. Luck of the draw, unfortunately.

closed 2022-09-27 by pfeerick

j-brubaker, 2022-10-19:
I was able to bench-test 2.8.0-RC3 on my backup Horus.  Setup: Horus X10S ACCST, Ghost Module

1. Verified 2.7 flashes telemetry values in red, audible telemetry warnings, and even had a 2sec failsafe
2. Flashed 2.8.0-RC3
3. When Sample mode: Normal, same behavior as 2.7
4. Change Sample mode to "OneBit", JR module reconnects (beeps), telemetry still bad
5. Reboot radio
6. Perfect telemetry, no red flashing, no audible warnings

Video:  [2.8.0-RC3 test](https://youtu.be/HYWUeEPpZ_0)

Aaaaaand as I'm typing this I have the radio on my bench, connected to the RX, and it just failsafed for about 5 seconds (radio announced "telemetry lost" and flight controller double-beeped).  This was maybe after 30 minutes or so.  Shoot.

pfeerick, 2022-10-20:
lol... I was going to say OneBit really should have no effect for Ghost -  channels will be sent at 400k, and telemetry at 115k - so if you're having trouble with a Horus X10S and Ghost I think are bigger problems. There is not even a need for the inverter mod, as the speed limit is only for data going in, not out, so slow inverters aren't an issue. Interesting though that it was able to mask the issue for a while... very interesting! Not what you would want to find out in flight though 😭