EdgeTX/edgetx #1195: External device firmware update: present a dialog with device info

wimalopaan, 2021-11-29:
Before actually flashing the external device, present a dialog to the user with some info about the found device like:

- device type
- firmware version on the device
- ...

If the user can cancel this dialog, this would be a way to query some info about the external device

pfeerick, 2021-11-29:
Next time, can you also include key information such as `FrSkyFirmwareInformation` since this is what lead to the issue being created? Less having to re-invent the wheel. 😉

If you look more closely at any of the code mentioning `FrSkyFirmwareInformation`, you'll quickly realise this is a data structure used to pull information out of the firmware update FILE, *not* from the receiver. 

Is this related to SPORT_VERSION_ACK, SPORT_VERSION_REQ?

wimalopaan, 2021-11-30:
Sorry, this one was created in a hurry ...

You mean this state machine?

https://github.com/EdgeTX/edgetx/blob/d3f35e83343f7d45048af76c01fa875970d4b733/radio/src/io/frsky_firmware_update.cpp#L86-L91

pfeerick, 2021-11-30:
More a "what makes you think we can get the firmware version of the device" question... as AFAIK, given what the OTX devs have said, at least the ACCST series of receiver have no mechanism for coughing up their firmware version. i.e. the only way to know is to flash it with a set version. 

That's why I was wondering if this was related to that code you linked above, meaning it is probably only for SPort stuff, and the fun will be finding out what stuff works. 😁

wimalopaan, 2021-11-30:
Well, I stumpled upon the comment in the above state-machine.
Ok, if there is no chance, I'll close this issue.

closed 2021-11-30 by wimalopaan

pfeerick, 2021-11-30:
I don't know about *no* chance - since that that packet/frame inspector does suggest that a `SPORT_VERSION_ACK` frame does exist - but as I said, it would be a matter of finding out how to use it / what supports it. More investigation is needed - especially since Frsky like to maintain a black box on matters such as this.

wimalopaan, 2021-11-30:
Maybe, I have the time to attach the LA to it and what the packets are. But a little information of FrSky or Radiomaster would be helpful. Radiomaster is official sponsor of theis project, so there should be a chance to get this information a the first place.

pfeerick, 2021-11-30:
I can certainly ask, but I wouldn't hold my breath on that one... I'd ask you to cast your mind back to *why* there were issues with some receivers and Frsky... and they were withdrawn from sale ;)

wimalopaan, 2021-11-30:
Radiomaster still sells the smaller RXs like R81/84/86/88 ... they should want to give an answer, or should disclose the full protocoll spec to the developers. At least if the plan to ship their radios with EdgeTx (or OpenTx).

pfeerick, 2021-11-30:
No, it doesn't work like that. Those receivers are all D8 "compatible", do not have firmware updates, nor mention any support for firmware updating in the manuals. :-P So really have nothing to do with EdgeTX or OpenTX... since the RF link side is up to the MPM / module to manage. As far as disclosing the protocol information, it's not really up to them to disclose it since it's Frsky's protocol, not theirs.

wimalopaan, 2021-11-30:
Ok, I wasn't aware of the fact, that their smaller RX aren't updateble.
And the R168 was discontinued due to legal issue with FrSky, because they hijacked the FrSky RX FW and put that on their R168? Then it is clear to me, the they are not in the position give any details on that. Sad ...

BTW: I do not mean the OTA update, only connecting the RX via S.Port to the radio

pfeerick, 2021-11-30:
Yup, since they only made compatible hardware, and it intruded on the D16 space, so Frsky wasn't happy with them or Jumper. They probably know nothing about the protocol or firmware, but it won't hurt to ask. Their smaller receivers may be updatable - they just don't make any mention of it, and whilst there is a heading for firmware under downloads it is empty. There could also simply be no real reason to since D8 doesn't need FCC/EU/v1/v2 variants because there are none AFAIK 😆 . 

Yeah, OTA update I know nothing about, nor have any hardware that supports it :grin:

wimalopaan, 2021-11-30:
Ok, then I would like to kindly ask you to contact RM on that ;-) 

In general, this would make very much unexperienced users happy, to know what their RXs are about and to show them a dialog with the information.