EdgeTX/edgetx #1197: Binding with or without telemetry

Torbjorn1305, 2021-12-01:
Hi
In nightlies I now see when binding, the question about channel and telemtry is the same as it is in Open-Tx.
Binding without telemtry is ok channel 1-8, but in channel 9-16 the telemtry is still on.

Best regards
Torbjorn1305

raphaelcoeffic, 2021-12-01:
The function looks ok: https://github.com/EdgeTX/edgetx/blob/d94b6fb9e83992866b6451ddc5515f3daab637ca/radio/src/gui/colorlcd/model_setup.cpp#L182

are you sure you chose the very last option?

wimalopaan, 2021-12-01:
For me actual main works like expected

Torbjorn1305, 2021-12-01:
HiI use verson: 2.6.0 (d3f35e83) and I trying rigt now and it did not work. Torbjørn JohnsenHågen 268522 BeisfjordNorge Fra: Raphael CoefficSendt: onsdag 1. desember 2021 kl. 16:01Til: EdgeTX/edgetxKopi: Torbjorn1305; AuthorEmne: Re: [EdgeTX/edgetx] Binding with or without telemetry (Issue #1197) The function looks ok: https://github.com/EdgeTX/edgetx/blob/d94b6fb9e83992866b6451ddc5515f3daab637ca/radio/src/gui/colorlcd/model_setup.cpp#L182are you sure you chose the very last option?
	
      
		Virusfri. www.avast.com

raphaelcoeffic, 2021-12-02:
> HiI use verson: 2.6.0 ([d3f35e8](https://github.com/EdgeTX/edgetx/commit/d3f35e83343f7d45048af76c01fa875970d4b733)) and I trying rigt now and it did not work. Torbjørn

Ok, let’s verify we all have the same understanding:
- which module is used? Internal or external? What type exactly?
- When you say “it does not work”, what do you mean exactly? Telemetry packets are still coming? Which sensors exactly do you see?
- Do you test only with the one receiver & module turned ON and connected, or is there anything else in the mix?

Torbjorn1305, 2021-12-02:
Fra: Torbjørn JohnsenSendt: torsdag 2. desember 2021 kl. 11:43Til: EdgeTX/edgetxEmne: SV: [EdgeTX/edgetx] Binding with or without telemetry (Issue #1197) I use Radiomaster TX16S radio and internal modul V1.3.3.0 AETR. I have to bind two receiver in redundansy system, and one of them have to be bind without telemetry. The best way doing it are: Bind primary receiver channal 1-8 with telemetry on and bind secondare receiver channal 9-16 without telemetry.I have tried to bind 1-8 without telemtry, and this is is Ok but 9-16 is not Ok.I have tried different receivers, and both I have tried is the same. Frsky RX6R and Frsky RX8R. Both of them show A2 ID 250, RSSI and RxBt ID 248.I don`t understand it`s working when I use 1-8 without telemetry but not With 9-16 without telemetry.The two receivers are connected together with s-bus.When i Bind 1-8 without telemetr no sensors displays. And I think 9-16 have to be the same.You can see the Pictures. Binding 1-8 without telemetry show RSSI 0 db. Binding 9-16 without telemetry show RSSI 111 db.  Torbjørn JohnsenHågen 268522 BeisfjordNorge Fra: Raphael CoefficSendt: torsdag 2. desember 2021 kl. 08:11Til: EdgeTX/edgetxKopi: Torbjorn1305; AuthorEmne: Re: [EdgeTX/edgetx] Binding with or without telemetry (Issue #1197) HiI use verson: 2.6.0 (d3f35e8) and I trying rigt now and it did not work. TorbjørnOk, let’s verify we all have the same understanding:which module is used? Internal or external? What type exactly?When you say “it does not work”, what do you mean exactly? Telemetry packets are still coming? Which sensors exactly do you see?Do you test only with the one receiver & module turned ON and connected, or is there anything else in the mix?

raphaelcoeffic, 2021-12-02:
@Torbjorn1305 but you are binding each receiver with a different module, right? What is the other module?

pfeerick, 2021-12-02:
What is the reason for the SBUS connection? Is this what allows the redundancy? Or are you [using this as well](https://www.frsky-rc.com/product/redundancy-bus/)? As I did try two X8R a few weels ago, one with telemetry, and one without, and it appeared to be working correctly. But this was without the SBUS connection.

Torbjorn1305, 2021-12-03:
HiNo the receivers are bind in same modul and same number because they have to work in redundansy mode.The S-bus Connection is for safaty. If one of the receiver lost signal the other one takes over. Only one of the receivers are connected with servos. That is redundansy.Only one of the receivers can be bind with telemetry because the system in the radio can`t have telemetry signal from two different receivers at the same time.I give you a link from youtube, maybe you can see the system and understand better than I can write it.https://www.youtube.com/watch?v=oFTqVhBi74M&t=128s   Torbjørn JohnsenHågen 268522 BeisfjordNorge Fra: Raphael CoefficSendt: torsdag 2. desember 2021 kl. 13:42Til: EdgeTX/edgetxKopi: Torbjorn1305; MentionEmne: Re: [EdgeTX/edgetx] Binding with or without telemetry (Issue ***@***.*** but you are binding each receiver with a different module, right? What is the other module?
	
      
		Virusfri. www.avast.com

pfeerick, 2021-12-04:
Oh, I didn't realise that was what the R was about! Thank you, interesting!

Right, so I just tried both `CH9-16 Telem ON` and `CH9-16 Telem OFF` again with X8R, and it worked as expected - telemetry worked the first time, but not the second. Are you fitting the jumper between signal pins 1&2 when binding with Telem OFF? This _is_ required, as it tells the receiver what mode to bind in. If you don't fit the jumper, regardless of what you set on the transmitter side, the receiver will bind with telemetry on.

Torbjorn1305, 2021-12-05:
HiX8R. The «R» in this receiver means not Redundanse.RX8R is an redundancy receiver.All the FRSKY receivers who start With an «R» in the name is an Redundancy receiver.You have the one and only receiver from FRSKY who need jumper to bind whithout telemetry or I have never use jumper on X8R.Every, yes I mean every other receiver have no jumper in the discription for binding.So no I know you never gone see what I mean about the differens between 1-8 telemetry OFF and 9-16 telemetry OFF. It have to be the same but you can`t see that.I have receivers «RX6R, S8R, S6R, RX8R, X8R and RXSR». All of them I have bind in Open-Tx channel 1-8 and 9-16 telemetry OFF.I do the same in EDGE-Tx channel 1-8 telemetry Off. Why not in 9-16 telemtry off?. That is the only problem. Torbjørn JohnsenHågen 268522 BeisfjordNorge Fra: Peter FeerickSendt: lørdag 4. desember 2021 kl. 05:03Til: EdgeTX/edgetxKopi: Torbjorn1305; MentionEmne: Re: [EdgeTX/edgetx] Binding with or without telemetry (Issue #1197) Oh, I didn't realise that was what the R was about! Thank you, interesting!Right, so I just tried both CH9-16 Telem ON and CH9-16 Telem OFF again with X8R, and it worked as expected - telemetry worked the first time, but not the second. Are you fitting the jumper between signal pins 1&2 when binding with Telem OFF? This is required, as it tells the receiver what mode to bind in. If you don't fit the jumper, regardless of what you set on the transmitter side, the receiver will bind with telemetry on.
	
      
		Virusfri. www.avast.com

pfeerick, 2021-12-05:
Hm, really? The only one? The instructions for this are exactly the same as the X8R. 

_However_, I tried this on OTX also, and I see what you mean now. Whilst the instructions reference the X12S as not *requiring* the jumpers to set, it seems _any_ of the radios that present the bind time options should allow you to bind in those modes without the jumpers being set. On OTX, I was able to bind in all four modes without a jumper, but on ETX the fourth option, the `9-16 TELEM OFF` did not function correctly - it essentially bound the same as `9-16 TELEM ON`. However, it seems the jumpers override what is set on the transmitter, which is why this worked. 

[image]

pfeerick, 2021-12-05:
I think I see the culprit, so will test it and a fix in shortly if that is the case.

pfeerick, 2021-12-05:
And it is. Thanks for reporting it and hanging in there! :)

closed 2021-12-05 by pfeerick

Torbjorn1305, 2021-12-08:
Hi It is possible to use frsky R9M module placed in Radiomaster TX16S With ETX? Torbjørn JohnsenHågen 268522 BeisfjordNorge Fra: Peter FeerickSendt: mandag 6. desember 2021 kl. 00:37Til: EdgeTX/edgetxKopi: Torbjorn1305; MentionEmne: Re: [EdgeTX/edgetx] Binding with or without telemetry (Issue #1197) And it is. Thanks for reporting it and hanging in there! :)
	
      
		Virusfri. www.avast.com

pfeerick, 2021-12-09:
Try it and find out if you have one! R9M is in the list of external RF options on the TX16. There are some issues with it still that are being tracked in other issues, but it is somewhat working.

commit 0618d6ef (2021-12-05): [colorlcd] Fix Frsky 9-16 TELEM OFF not working
Resolves #1197