EdgeTX/edgetx #1198: Model notes doesn't read model.txt

tonims1, 2021-12-01:
Since the implementation of yawl It doesn't read the .txt model. read the .yml model

RM TX16S

Nightlies 2.6
[image]

eshifri, 2021-12-01:
I cannot reproduce it.
I assume you have checked the model6.txt or Ultron1.txt file is still present in the Models folder?

tonims1, 2021-12-01:
Yes. in the models folder I have the .txt and .yml files. but when I click to read the txt model It read the yml model

eshifri, 2021-12-01:
Can you please clarify?
If you go into the Radio Settings->SD card->Models, click model6.txt, select Read from the menu model6.yml file opens?

tonims1, 2021-12-01:
no. if I go to Radio Settings-> SD card-> Models, click model6.txt It reads well. but if I do it from the main screen click model notes its show me the yml fi
[image]
le

eshifri, 2021-12-01:
Aha. I still cannot reproduce it.
When you start the radio or change the model, do you see the same problem?
(Sorry for so many questions - I'm just trying to put my finger on the place in the code where the problem might be.)
It is all very strange since the text viewer does not consider yml to be a text format.

tonims1, 2021-12-01:
Yes it happens in all models . when I want to read the model notes from the main screen always read model .yml

eshifri, 2021-12-01:
My question was a little bit different: when you change the model and the modelX.txt file is present for a newly selected model, it should be displayed. Do you see the YML file in this case as well?

tonims1, 2021-12-01:
yes

tonims1, 2021-12-01:
https://user-images.githubusercontent.com/7953406/144309249-f4f4433c-65b0-4094-8fb7-7b9238f1aaf8.mp4

eshifri, 2021-12-01:
Thank you! 
I know what the problem is. Will fix it shortly.

closed 2021-12-02 by pfeerick

tonims1, 2021-12-02:
Thank you for your attention

PR #1202 (2021-12-02): use yml instead of bin to construct notes file name from model file n…
Fixes #1198 

Summary of changes: when model notes file is named modelX.txt we should assume that the model file has "yml" rather than "bin" extension.