EdgeTX/edgetx #1199: compiled in spanish have a line above

tonims1, 2021-12-01:
compiled in Spanish all the text comes out with a line above it

RM TX16S

2.5 , 2.6
[image]
.
[image]
[image]

richardclli, 2021-12-07:
Interesting, let me take a look first.

raphaelcoeffic, 2021-12-10:
> Interesting, let me take a look first.

I could bet this has something to do with the generated font that might have some black parts in it (maybe while extending the height for some special character, or whatever).

richardclli, 2021-12-17:
font_9, font_15, font_16, font_bold16 has problems, need to figure out why.

richardclli, 2021-12-18:
The problem is due to some y_offset is negative (ES is the case).  What I did is whenever y_offset is -ve, I will try to enlarge the image first and offset the y_offset back to 0, so that the final crop will not cause a black line.

PR #1268 (2021-12-27): Resolve a problem in font generation when y_offset is negative.
Fixes #1199

Summary of changes:
- fixed a problem in font bitmap generation when y_offset is negative
- fixed missing special char because of hardcoded list (use dynamic list instead just like cjk) only for color radios
- isolated encoding method for B/W radios
- fixed B/W radio encoding based on the fixed fonts
closed 2021-12-27 by pfeerick